adafruit/Adafruit_Blinka #520: install script bricks zero

Fredmf, 2021-10-08:
i did set up a fresh sd, seet up usb gadget mode and internet, all prepared, running the raspi-blinka installscript bricked the thing, now all i get over lsusb is a ID 0a5c:2763 Broadcom Corp. BCM2708 Boot device when plugging in the pi...

micglin, 2021-11-14:
(hope this doesn't re-open the issue) FYI @Fredmf, I've too have had (what I assume to be) the kernel update toast a fresh Raspberry Pi OS install on Pi Zeros a handful of times.  Most commonly, within minutes of 1st boot, during the initial "sudo apt update && sudo apt upgrade -y" phase.  Doesn't usually reproduce itself, so while it's annoying to reburn the IMG and start over, it's only a few minutes lost.

makermelissa, 2021-11-15:
@micglin if you're referring running https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/adafruit-pitft.py, then I think the issue is it's not displaying a prompt to update the files for some reason. I usually just hit y and enter and it seems to work. If that's the case, please open an issue at https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/issues.

micglin, 2021-11-15:
@makermelissa AH, sorry for the misunderstanding.  No, not specifically while running the raspi-blinka.py script, I meant during the very first "apt upgrade" on Pi Zero boards in general.  

With Pi0 boards, I always copy a "ssh" and "wpa_supplicant.conf" file to the /boot volume after imaging, and prefer to start and stick with SSH access.  As such, my first command is to pull my keys to .ssh/authorized_keys, my second is to apt update/upgrade, 3rd is reboot.  So a board being unavailable because WiFi doesn't come back up is a fluke issue I've encountered occasionally, and was trying to reinforce to @Fredmf is a fluke happening with Raspberry Pi OS, and not indicative of a bug or issue with the Adafruit script, which both you and ladyada mention simply runs "apt-get upgrade" as seen here:
https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/raspi-blinka.py#L42

makermelissa, 2021-11-15:
You can use the Raspberry Pi Image and press Control+Shift+X to get to advanced options and have it automatically add that stuff to the board. I used to use your method, but the image is much easier.