adafruit/Adafruit_Blinka #523: Add support for raspberry pi zero w 2

lexcyn, 2021-11-03:
The module adafruit-platformdetect for blinka does not detect the raspberry pi zero w 2 and thus python scripts that rely on this fail to run.

caternuson, 2021-11-03:
Pi Zero 2 W support was added here:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/193
and released in 3.17.0:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/releases/tag/3.17.0

The install process covered here:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi
was tested on a Pi Zero 2 W starting with the 2021/05/07 release of Raspberry Pi OS Lite and worked.

What version of PlatformDetect are you using?

lexcyn, 2021-11-04:
Interesting. I am running 3.17 however when running my script, it throws this error (note how it said 3.153 was running. I noticed that the version in dist-packages was different than site-packages, so I figure something in the install was corrupt and pulling the wrong one - see my final fix below):

```
File "/usr/local/lib/python3.7/dist-packages/board.py", line 264, in <module>
    platform.system(), package[0], package[1]
NotImplementedError: Adafruit-PlatformDetect version 3.15.3 was unable to identify the board and/or microcontroller running the Linux platform. Please be sure you have the latest packages running: 'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'
```

I ended up running a `sudo python3 -m pip install --upgrade --force-reinstall adafruit-blinka Adafruit-PlatformDetect` and that looks like it fixed the issue. Not sure what was going on, thanks for checking into it so quickly!

closed 2021-11-04 by lexcyn