adafruit/Adafruit_Blinka #525: Support for BeagleBone Green Gateway

keenanjohnson, 2021-11-07:
I see that the [Beaglebone Green Gateway](https://www.seeedstudio.com/SeeedStudio-BeagleBone-Green-Gateway-p-4586.html) is not currently supported by this library. I think it might be nice for this board to be supported as it has some great features!

Does it seem reasonable to support this board here?

keenanjohnson, 2021-12-06:
It seems straightforward to add this to the board.py file since the BEAGLEBONE_GREEN is already supported and the pin assignments will be the same.

https://github.com/adafruit/Adafruit_Blinka/blob/f8c28157c2ffceaa436f106f0bbc4678b889cc39/src/board.py#L82-L83

I'm going to go for it!

makermelissa, 2021-12-06:
Fixed via #529

closed 2021-12-06 by makermelissa