adafruit/Adafruit_Blinka #534: Freeze when using I2C devices and u2if.

domdfcoding, 2021-12-28:
Hi,

I'm trying to use Blinka with a Qt Py RP2040 running u2if. I can make the NeoPixel flash without issue, but when trying to use an I2C device Python freezes at the point of setting up the I2C bus. I can't quit with Ctrl+C as normal; instead I have to use Ctrl+\ (SIGQUIT) or disconnect the Qt Py.

I've tried with both a DS3502 breakout and the Wii Nunchuk breakout, but the results are the same.

The code I'm trying to use for the nunchuk is:

```python
import board
import adafruit_nunchuk

print("Ready")
nc = adafruit_nunchuk.Nunchuk(board.I2C())
```
and with the DS3502:

```python
import board
import adafruit_ds3502

print("Ready")
ds3502 = adafruit_ds3502.DS3502(board.I2C())
```

but in both cases it gets stuck after printing "Ready".

If I disconnect the board the traceback is as follows:

<details><summary>Traceback</summary>

```python
Traceback (most recent call last):
  File "code.py", line 9, in <module>
    nc = adafruit_nunchuk.Nunchuk(board.I2C())
  File "adafruit_nunchuk.py", line 65, in __init__
    i2c_dev.write(b"\xF0\x55")
  File "venv/lib/python3.8/site-packages/adafruit_bus_device/i2c_device.py", line 84, in write
    self.i2c.writeto(self.device_address, buf, start=start, end=end)
  File "venv/lib/python3.8/site-packages/busio.py", line 165, in writeto
    return self._i2c.writeto(address, memoryview(buffer)[start:end], stop=stop)
  File "venv/lib/python3.8/site-packages/adafruit_blinka/microcontroller/rp2040_u2if/i2c.py", line 22, in writeto
    rp2040_u2if.i2c_writeto(address, buffer, start=start, end=end)
  File "venv/lib/python3.8/site-packages/adafruit_blinka/microcontroller/rp2040_u2if/rp2040_u2if.py", line 260, in i2c_writeto
    self._i2c_write(address, buffer, start, end)
  File "venv/lib/python3.8/site-packages/adafruit_blinka/microcontroller/rp2040_u2if/rp2040_u2if.py", line 229, in _i2c_write
    resp = self._hid_xfer(
  File "venv/lib/python3.8/site-packages/adafruit_blinka/microcontroller/rp2040_u2if/rp2040_u2if.py", line 89, in _hid_xfer
    return self._hid.read(64)
  File "hid.pyx", line 200, in hid.device.read
OSError: read error
```
</details>

ladyada, 2021-12-28:
@caternuson is out for holiday break, will take a look when they return!

caternuson, 2021-12-29:
Can't recreated this locally.

Can you post complete minimal code listing that demonstrates this issue. The callback and code you've provided don't agree.

caternuson, 2022-01-10:
Closing due to lack of response. Can reopen if problem can be recreated with complete example code.

closed 2022-01-10 by caternuson