adafruit/Adafruit_Blinka #535: `import board` with MCP2221 fails with `OSError: open failed`

ezio-melotti, 2021-12-29:
I spent a couple of nights trying to read data from a SCD30 connected to a Linux machine through a MCP2221 following [this example](https://circuitpython.readthedocs.io/projects/scd30/en/latest/#usage-example), but I was getting a traceback during the import of the `board` module.  To run the MCP2221 I followed [this article that explains how to install all the necessary packages and also how to blacklist and replace the native hid_mcp2221 Linux driver](https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/linux) (all the [post-install checks](https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/post-install-checks) were successful).

Eventually I narrowed it down to a minimal set of instructions to reproduce the failures consistently.

The first error I got was 
```py
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/board.py", line 288, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported GENERIC_LINUX_PC
```
I was able to fix this by restarting Python, setting the `BLINKA_MCP2221` environment variable to `1`, and trying `import board` again.  However, I still got an error:
```py
>>> import os; os.environ['BLINKA_MCP2221'] = '1'
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/board.py", line 197, in <module>
    from adafruit_blinka.board.microchip_mcp2221 import *
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/board/microchip_mcp2221.py", line 2, in <module>
    from adafruit_blinka.microcontroller.mcp2221 import pin
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/pin.py", line 2, in <module>
    from .mcp2221 import mcp2221
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 386, in <module>
    mcp2221 = MCP2221()
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 55, in __init__
    self._reset()
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 131, in _reset
    raise OSError("open failed")
OSError: open failed
>>> 
```
If I try again immediately after, I get a different error:
```py
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/board.py", line 197, in <module>
    from adafruit_blinka.board.microchip_mcp2221 import *
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/board/microchip_mcp2221.py", line 2, in <module>
    from adafruit_blinka.microcontroller.mcp2221 import pin
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/pin.py", line 2, in <module>
    from .mcp2221 import mcp2221
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 386, in <module>
    mcp2221 = MCP2221()
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 53, in __init__
    self._hid.open(MCP2221.VID, MCP2221.PID)
  File "hid.pyx", line 113, in hid.device.open
OSError: open failed
```

In both cases, the error comes from a `self._hid.open(MCP2221.VID, MCP2221.PID)`, and it happens whenever I try to open a device that is already open.  By looking at the code, the `__init__` of the `MCP2221` class does:
https://github.com/adafruit/Adafruit_Blinka/blob/1ac53c86297e869220ea41d8740f01852609dd65/src/adafruit_blinka/microcontroller/mcp2221/mcp2221.py#L51-L55
and the `_reset` method that it calls does (notice the other `open` at line 125):
https://github.com/adafruit/Adafruit_Blinka/blob/1ac53c86297e869220ea41d8740f01852609dd65/src/adafruit_blinka/microcontroller/mcp2221/mcp2221.py#L119-L131

It seems that the first traceback is caused by the fact that the `__init__` successfully opens the device (at line 53), but then calls `self._reset()` at line 55, that tries to open the  -- already opened -- device again (at line 125), causing the traceback.  The second traceback can be explained by the fact that after the first error, the `__init__` doesn't clean up after itself and doesn't call `self._hid.close()`, so the device is still open when the `self._hid.open(MCP2221.VID, MCP2221.PID)` at line 53 is executed.  However after waiting for a while, the `import board` fails again in the `_reset()` method (maybe the device is automatically closed -- perhaps after being garbage collected -- and the `open` at line 53 is successful again).

I verified that trying to open an already open device (like `_reset` does after the `__init__`, or after two consecutive `import board`) results in an error:
```py
>>> import hid; h = hid.device(); h.open(0x04D8, 0x00DD);
>>> h.open(0x04D8, 0x00DD)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "hid.pyx", line 113, in hid.device.open
OSError: open failed
```

[... 57 lines cut ...]
This should fix the slow `.close()` and some other issues, but probably not the import error.  To fix the import error, I eventually used this solution:
```py
>>> import os; os.environ['BLINKA_MCP2221'] = '1'
>>> import os; os.environ['BLINKA_MCP2221_RESET_DELAY'] = '20'
>>> import board
>>>
python3: io.c:2115: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.
Aborted (core dumped)
```
This is the only way I found to successfully `import board`, and after that I was able to run the example and read data from the SCD30 sensor (even though it still crashes upon exit).

There are still a few things that are not clear to me:
1. Why is the call to `self._reset()` necessary and what does it do?
2. Why does `_reset` open the device again, and why does it work after adding a 20s delay?
3. What exactly is altering the state of the MCP2221 and causing `.close()` to become slow?
4. Why on my machine a 20s delay is necessary to be able to `import board`?
5. Why does [the linked article](https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/linux) requires the native Linux driver to be blacklisted?

#243 and #244 seems related.

---

Tl;DR:
* Trying to `import board` with `BLINKA_MCP2221 == 1` (and importing the `MCP2221` class) fails unless a 20s delay is added.
* The `MCP2221` class/module leave the MCP2221 in an altered state in case of error or even when closing the program after a successful import.

caternuson, 2021-12-29:
Setting the environment variable is required. The guide covers that.

What specific linux distro are you running?

ezio-melotti, 2021-12-29:
I'm using Kubuntu 21.10.

caternuson, 2021-12-29:
This behavior is repeatable after a reboot? Reboot, plug in MCP2221, set envar, launch python, import board - and you get a traceback?

ezio-melotti, 2021-12-29:
After removing and blacklisting the native MCP2221 driver I did reboot but I didn't unplug the MCP2221.  I now tried to unplug it, reboot, plug it again, launch Python with `$ BLINKA_MCP2221=1 python3` and then `import board` worked fine, with no tracebacks and no delays.  I also tried to launch Python normally and setting the env var with `os.environ['BLINKA_MCP2221'] = '1'` and `import board` still worked.

However in both cases it still crashed when I closed Python with ctrl+d:
```py
>>> import board
>>> 
python3: io.c:2115: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.
Aborted (core dumped)
```

Do you think the problem was caused by the fact that I didn't plug the MCP2221 after starting up the PC, but left it plugged in instead?

FWIW I also noticed that every time I `import board`, I get a new entry in `dmesg`, with a new increasing device number:
```
[  691.780257] usb 5-1: USB disconnect, device number 5
[  692.235157] usb 5-1: new full-speed USB device number 6 using xhci_hcd
[  692.435156] usb 5-1: New USB device found, idVendor=04d8, idProduct=00dd, bcdDevice= 1.00
[  692.435169] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  692.435174] usb 5-1: Product: MCP2221 USB-I2C/UART Combo
[  692.435178] usb 5-1: Manufacturer: Microchip Technology Inc.
[  692.451298] cdc_acm 5-1:1.0: ttyACM0: USB ACM device
[  692.459950] hid-generic 0003:04D8:00DD.0006: hiddev0,hidraw1: USB HID v1.11 Device [Microchip Technology Inc. MCP2221 USB-I2C/UART Combo] on usb-0000:00:10.0-1/input2
```

caternuson, 2021-12-29:
Calling `import board` results in a call to resetting the board, which is what is causing the dmesg traffic.

Unknown what the original issue was, but seems possible things just got in bad state while trying to install everything, set the envar, etc. Something from the failed attempts hadn't shut down fully and was interfering with the follow on attempts.

ezio-melotti, 2021-12-30:
I did more tests.  With the `import board` working, I tried to unplug the MCP2221 + SCD30 and plug it in again, and it still worked.

I then tried to restart the PC without unplugging it, and after the reboot I got the usual traceback:
```py
>>> import os; os.environ['BLINKA_MCP2221'] = '1'
>>> import board
Traceback (most recent call last):
 File "<stdin>", line 1, in <module>
   ...
 File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 55, in __init__
   self._reset()
 File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 131, in _reset
   raise OSError("open failed")
OSError: open failed
```

`dmesg` shows this shortly after boot:
```
[    2.258811] usb 5-1: New USB device found, idVendor=04d8, idProduct=00dd, bcdDevice= 1.00
[    2.258823] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.258828] usb 5-1: Product: MCP2221 USB-I2C/UART Combo
[    2.258832] usb 5-1: Manufacturer: Microchip Technology Inc.
[    2.279213] hid-generic 0003:04D8:00DD.0002: hiddev0,hidraw1: USB HID v1.11 Device [Microchip Technology Inc. MCP2221 USB-I2C/UART Combo] on usb-0000:00:10.0-1/input2
[    2.282744] mcp2221 0003:04D8:00DD.0002: hidraw1: USB HID v1.11 Device [Microchip Technology Inc. MCP2221 USB-I2C/UART Combo] on usb-0000:00:10.0-1/input2
```
and this after the attempted `import board`:
```
[   87.356930] usb 5-1: USB disconnect, device number 2
[   87.811386] usb 5-1: new full-speed USB device number 3 using xhci_hcd
[   88.010825] usb 5-1: New USB device found, idVendor=04d8, idProduct=00dd, bcdDevice= 1.00
[   88.010833] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   88.010835] usb 5-1: Product: MCP2221 USB-I2C/UART Combo
[   88.010837] usb 5-1: Manufacturer: Microchip Technology Inc.
[   88.025960] cdc_acm 5-1:1.0: ttyACM0: USB ACM device
[   88.034220] mcp2221 0003:04D8:00DD.0003: hidraw1: USB HID v1.11 Device [Microchip Technology Inc. MCP2221 USB-I2C/UART Combo] on usb-0000:00:10.0-1/input2
```
It seems the call to `_reset` was successful (`dmesg` shows the USB disconnected and reconnected), but then the `self._hid.open()` in `_reset()` failed.  Subsequent attempts at importing `board` all fail.

I also noticed that despite the `hid_mcp2221` module being blacklisted:
```sh
$ cat /etc/modprobe.d/blacklist.conf | tail -n1
blacklist hid_mcp2221
```
it still showed up in `dmesg` and in `lsmod`:
```sh
$ sudo dmesg
...
[   88.034220] mcp2221 0003:04D8:00DD.0003: hidraw1: USB HID v1.11 Device [Microchip Technology Inc. MCP2221 USB-I2C/UART Combo] on usb-0000:00:10.0-1/input2
$ lsmod | grep hid
mac_hid                16384  0
hid_mcp2221            20480  0
hid_generic            16384  0
usbhid                 61440  0
hid                   135168  3 usbhid,hid_generic,hid_mcp2221
```
I tried to remove it, and after that I was able to `import board`:
```
$ sudo rmmod hid_mcp2221
$ lsmod | grep hid
mac_hid                16384  0
hid_generic            16384  0
usbhid                 61440  0
hid                   135168  2 usbhid,hid_generic
```
`dmesg` seems to provide some insight on what's going on:
```
$ sudo dmesg
...
[  469.223375] usb 5-1: USB disconnect, device number 4
[  469.678659] usb 5-1: new full-speed USB device number 5 using xhci_hcd
[... 34 lines cut ...]
mac_hid                16384  0
hid_generic            16384  0
usbhid                 61440  0
hid                   135168  2 usbhid,hid_generic
```

After this I'm able to start my PC with the MCP2221 and SCD30 plugged in and `import board` directly without issues.  However it still crashes consistently upon exit (also verified on another Linux machine):
```py
>>> import os; os.environ['BLINKA_MCP2221'] = '1'
>>> import board
>>> 
Segmentation fault (core dumped)
```

Should I create a separate issue for that?

---

TL;DR:
* The native `hid_mcp2221` module causes `import board` to fail with `OSError: open failed`.
*  If `dmesg` shows `mcp2221 0003:04D8:00DD.0006: hidraw1: USB HID v1.11 Device [Microchip Technology Inc. MCP2221 USB-I2C/UART Combo]`, then the `hid_mcp2221` module is being used.
* Removing it with `rmmod hid_mcp2221` temporarily fixes the issue, but the `hid_mcp2221` module gets reloaded automatically, and the following imports will fail again.
* Adding `blacklist hid_mcp2221` to `/etc/modprobe.d/blacklist.conf` prevents the automatic reload, but the module still gets loaded at startup.
* Running `sudo update-initramfs -u` ensures that the blacklisting is picked up after a reboot.
* Leaving everything plugged in during reboot or unplugging it and plugging it back in doesn't seem to affect the outcome, as long as the `hid_mcp2221` module is blacklisted.

caternuson, 2021-12-30:
> Running sudo update-initramfs -u ensures that the blacklisting is picked up after a reboot.

This seems to be the key step. Thanks for all the testing and info. This additional step should be added to the [linux setup for the MCP2221 guide](https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/linux).

for ref:
https://wiki.debian.org/KernelModuleBlacklisting
```
1. Create a file '/etc/modprobe.d/<modulename>.conf' containing 'blacklist <modulename>'.
2. Run 'depmod -ae' as root
3. Recreate your initrd with 'update-initramfs -u'
```

caternuson, 2021-12-30:
@ezio-melotti Does this updated information seem OK?
https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/linux#remove-native-mcp2221-driver-3076488-18

ezio-melotti, 2021-12-31:
Thanks for updating the guide!

I have a couple of comments about the changes:
1. > The native driver can be temporarily removed (disabled) with `sudo rmmod hid_mcp2221`

   This should probably clarify that the driver is disabled until the device is plugged in again, and not for the whole session.  This also means that a simple `import board` might cause the `hid_mcp2221` driver to be reloaded, unless it's also blacklisted (I think this is because of the `self._reset()` that disconnects and reconnects the device, causing the `hid_mcp2221` driver to be loaded again).

   It would also be worth mentioning that removing `hid_mcp2221` will simply cause it to fall back on another native driver -- `hid_generic`.
2. > Then run the following command to also update the configuration in initramfs (details):

   This seems to work for Debian-based systems, but other distros (e.g. RedHat) don't seem to have/use `update-initframfs`.  https://unix.stackexchange.com/a/90692 lists some alternatives -- not sure if you want to link that thread and/or this issue in the guide.

Since I'm a bit wary of executing commands that I don't fully understand and/or that might affect the global configuration of my machine, it would be great if the guide expanded a bit on what some of these commands do and why they are necessary, in particular:
1. >  `sudo apt-get install libusb-1.0 libudev-dev`

   I initially skipped this command and it seems that -- at least on my system -- `libusb-1.0` was already installed (several other programs depend on it) and `libudev-dev` doesn't seem required (I now tried to remove it and everything seems to still work).  Including the command in the guide is probably useful though, since other people might not have `libusb-1.0` installed by default.  Perhaps the wording could be changed to something like "If you don't have `libusb-1.0` installed, install it with `sudo apt-get install libusb-1.0`.".
2. > Use a text editor to create and edit the file `/etc/udev/rules.d/99-mcp2221.rules` and add the following contents. `SUBSYSTEM=="usb", ATTRS{idVendor}=="04d8", ATTR{idProduct}=="00dd", MODE="0666"`

   While investigating I came across an issue (nonNoise/PyMCP2221A#16) that seems to imply that the rule is necessary in order to access the mcp2221 as a regular user, without having to use `sudo`.  After further research it seems that it specifically gives read/write access to everyone (`MODE="0666"`) for this specific USB device (`ATTRS{idVendor}=="04d8", ATTR{idProduct}=="00dd"`).

I also created #536 as a follow-up to this issue.

dalbabur, 2022-03-05:
Having the same issues. I have set `BLINKA_MCP2221 = 1` and `BLINKA_MCP2221_RESET_DELAY = 20`
I'm on a raspberry pi 4 with Bullseye. No `hid_mcp2221` driver was found (`rmmod: ERROR: Module hid_mcp2221 is not currently loaded
`), but I still blacklisted. 

I have tried all plug/reboot combinations.

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/board.py", line 182, in <module>
    from adafruit_blinka.board.microchip_mcp2221 import *
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/board/microchip_mcp2221.py", line 5, in <module>
    from adafruit_blinka.microcontroller.mcp2221 import pin
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/pin.py", line 5, in <module>
    from .mcp2221 import mcp2221
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 389, in <module>
    mcp2221 = MCP2221()
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 58, in __init__
    self._reset()
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/mcp2221/mcp2221.py", line 134, in _reset
    raise OSError("open failed")
OSError: open failed
```

ezio-melotti, 2022-03-05:
@dalbabur, can you paste the output of `lsmod | grep 'mcp\|hid'` and `sudo dmesg | grep -i mcp`?
If you install `hidapi` (`pip install hidapi`), open a Python interactive interpreter, and run:
```py
import hid
h = hid.device()
h.open(0x04D8, 0x00DD)
```
Do you get any error?

caternuson, 2022-03-05:
That's one of the post install checks:
https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/post-install-checks#check-that-mcp2221-can-be-found-3049740-7

Did you run that when setting things up initially?

dalbabur, 2022-03-08:
`lsmod` and `dmesg` commands dont return any output 

I did install `hidapi` and run the post-install checks without errors

caternuson, 2022-03-08:
Closing this issue. It's getting off track.

@ezio-melotti I think guide info has generally been updated per all the previous discussion above.

@dalbabur Please open a new issue. This seems to be something specific to your setup. Just tested setting up a MCP2221 up on a Pi4 and it generally worked OK. Also, just to point out - the only thing a MCP2221 is adding that a Pi does not already have natively are the ADCs. Pi's have I2C, SPI, UART, and tons of GPIO.

**BASIC HID TEST**
```python
(blinka) pi@raspberrypi:~ $ python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> hid.enumerate()
[{'path': b'1-1.3:1.2', 'vendor_id': 1240, 'product_id': 221, 'serial_number': '', 'release_number': 256, 'manufacturer_string': 'Microchip Technology Inc.', 'product_string': 'MCP2221 USB-I2C/UART Combo', 'usage_page': 0, 'usage': 0, 'interface_number': 2}]
>>> device = hid.device()
>>> device.open(0x04D8, 0x00DD)
>>> 
```
**BASIC BLINKA TEST**
```python
(blinka) pi@raspberrypi:~ $ python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.environ["BLINKA_MCP2221"]
'1'
>>> import board
>>> dir(board)
['G0', 'G1', 'G2', 'G3', 'I2C', 'SCL', 'SDA', '__blinka__', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__repo__', '__spec__', '__version__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']
>>>
```

closed 2022-03-08 by caternuson