adafruit/Adafruit_Blinka #537: is usb_hid planned?

werpu, 2022-01-01:
Hi I have a project on a raspberry pi zero2w where I try to emulate a usb hid (multiple in fact). Naturally I tried to use blinka only to find out that there is no usb_hid support yet. Sure there is a way to achieve this on lower level, but that would mean to generate or trigger all the needed usb identifier strings, which is an error prone process.
So my question/feature request. Is anything regarding usb_hid support planned. Sure this wont work for all raspberry pis but for some it will.

ladyada, 2022-01-01:
you want to have the raspi act as a usb keyboard to some other computer?

werpu, 2022-01-03:
Yes thats the plan... more like having a zero pi 2w act as a controller mouse keyboard combo on another computer.
I know there are microcontrollers which can do the job but in my use case i need a full blown pi.

makermelissa, 2022-01-03:
Ah, you must be referring to Serial Gadget, which I think only works on the older Pi Zeros like you mentioned. This would be a cool feature to have, but would require a bit more research to make sure it's even possible first.

ladyada, 2022-01-03:
yeah this is not on our todo list, check out pimoroni's key OS for something you can start with
https://learn.pimoroni.com/article/setting-up-the-keybow-os

makermelissa, 2022-01-03:
After a bit of discussion, this is not in the plans. If you would like to implement it yourself and submit a PR, we'd be happy to take a look at that.

closed 2022-01-03 by makermelissa