adafruit/Adafruit_Blinka #545: No SCL_1 on raspberry pi 3

StefanBaar, 2022-01-21:
Good day,

I am having problems using two I2C ports, SCL_1 and SDA_1 on  raspberry pi 3. 
 Only SCL and SDA appear to work... 

entering:
`i2c1 = busio.I2C(board.SCL_1, board.SDA_1, frequency=400000)`

gives:

`AttributeError: module 'board' has no attribute 'SCL_1'`

How can I use two SCL and SDA ports on Raspberry pi 3? 

I installed board via `pip3 install adafruit-blinka`.

Cheers

dhalbert, 2022-01-21:
See https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/using-i2c-or-spi-by-device-id for a special library that lets you use another I2C bus.

You are probably already aware of this, but you don't need a second bus for a second device unless that device has the same address as the first device. If they have different I2C addresses, they can share the same bus.

closed 2022-01-21 by dhalbert

StefanBaar, 2022-01-21:
thanks a lot. I will check it out.