adafruit/Adafruit_Blinka #546: Support GENERIC_LINUX_PC

septatrix, 2022-01-27:
It is not unusual for normal PCs to have GPIO and I2C interfaces. Places where these can be found are e.g. motherboards and VGA outputs but also RPi-style board not yet supported by this library. As this library already seems to have support for `adafruit_blinka/microcontroller/generic_linux` from by means of `Adafruit_Python_PureIO` it should not be hard to implement this.

Ideally this could even be used to unify the code for e.g. beaglebone and RPis though this would be only a nice side effect.

ladyada, 2022-01-27:
hiya - what PCs have gpio/i2c interfaces that are exposed to python?

septatrix, 2022-01-28:
For example my laptop has 19 i2c devices listed under `ls /dev/i2c-*` or `i2cdetect -l` of which the ones on the VGA output can be easily used with a VGA breakout

ladyada, 2022-01-28:
huh well thats nifty. most don't expose the i2c. if ya like you could PR a way to add access to these!

makermelissa, 2023-01-23:
With #647 added, you should be able to use the Adafruit_Python_Extended_Bus library for I2C.

closed 2023-01-23 by makermelissa