adafruit/Adafruit_Blinka #547: libgpiod_pulsein64 not found

jerryneedell, 2022-02-04:
I tried running test of pulseIn on a 64 Bit RaspiOS system and it complained that
libgpiod_pulsein64 is not found.

I  created my own
```
git clone https://github.com/adafruit/libgpiod_pulsein.git
cd libgpiod_pulsein/src
make 
sudo cp libgpiod_pulsein /usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein64
```
and it now works fine - even ran the dht22 simpletest!

However, I see that libgpiod_pulsein64 is in the repository but it is not installed via
sudo pip3 install adafruit-blinka

does this line have to be modified to include libgpiod_pulsein64?
https://github.com/adafruit/Adafruit_Blinka/blob/main/setup.py#L70

jerryneedell, 2022-02-05:
I verified that copying the existing libgpiod_pulsein64 from the repository to the folder created when I install adafruit blinka via pip3 works.  I tried to create a PR, but I am running into issue with pre-commit and I do not know how to test this locally. running sudo python3 setup.py install  does not work properly. 
I may just be simpler for someone who knows  what they are doing to exit line 70 of setup.py if that is the correct change.

jerryneedell, 2022-02-05:
It also appears to need to have libgpiod2 installed.
I installed libgpiod-dev and that installed libgpiod2 and libgpiod-dev

StevenAFC, 2022-02-06:
I am also having this issue, I have had to do the following in my Dockerfile:

`# Hack
RUN wget https://github.com/adafruit/Adafruit_Blinka/raw/main/src/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein64
RUN cp libgpiod_pulsein64 /opt/venv/lib/python3.10/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein64
RUN chmod u=rwx,g=rwx,o=rwx /opt/venv/lib/python3.10/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein64`

makermelissa, 2022-02-09:
Hmm, ok. I didn't realize Pip was not installing the binary, which is a problem. I'll take a look into if there's a way to remedy that. As for libgpiod-dev, we can update our raspberry pi blinka installer script, which might also be the answer for the missing binary.

makermelissa, 2022-02-10:
Ok, I noticed `libgpiod_pulsein` is being installed, so I think the issue has something to do with the executable file flag not being set. Should be an easy fix I think.