adafruit/Adafruit_Blinka #550: circuitpython_typing depends on numpy but numpy is not respected by setup.py

meandthemachine, 2022-02-11:
Hello,

I followed the [Adafruit guide for OLED displays](https://cdn-learn.adafruit.com/downloads/pdf/monochrome-oled-breakouts.pdf) and installed [adafruit/Adafruit_CircuitPython_SSD1306](https://github.com/adafruit/Adafruit_CircuitPython_SSD1306) system-wide via `sudo pip3 install adafruit-circuitpython-ssd1306`. That package depends on [Adafruit_Blinka](https://github.com/adafruit/Adafruit_Blinka).

After the installation finished I can't 
```
import adafruit_ssd1306
```
It says:
```
>>> import adafruit_ssd1306
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ssd1306.py", line 17, in <module>
    from adafruit_bus_device import i2c_device, spi_device
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 26, in <module>
    class I2CDevice:
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 65, in I2CDevice
    self, buf: WriteableBuffer, *, start: int = 0, end: Optional[int] = None
NameError: name 'WriteableBuffer' is not defined
```

Recently, type hinting was added to [Adafruit_CircuitPython_BusDevice](https://github.com/adafruit/Adafruit_CircuitPython_BusDevice) which therefore, in its `i2c_device.py`, now tries to import from `circuitpython_typing` which itself then depends on `numpy`. 

So I checked the dependencies. [Blinka's requirements.txt](https://github.com/adafruit/Adafruit_Blinka/blob/main/requirements.txt) lists numpy but this is not respected in its `setup.py`. So `pip3 install` says that all requirements are satisfied although numpy (and probably also binho-host-adapter) are missing:

```
$ sudo pip3 install Adafruit-Blinka
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: Adafruit-Blinka in /usr/local/lib/python3.7/dist-packages (6.20.2)
Requirement already satisfied: sysv-ipc>=1.1.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (1.1.0)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from Adafruit-Blinka) (0.7.0)
Requirement already satisfied: Adafruit-PlatformDetect>=3.13.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (3.19.5)
Requirement already satisfied: rpi-ws281x>=4.0.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (4.3.1)
Requirement already satisfied: pyftdi>=0.40.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (0.53.3)
Requirement already satisfied: Adafruit-PureIO>=1.1.7 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (1.1.9)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka) (1.2.1)
Requirement already satisfied: pyserial>=3.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka) (3.5)
$
```

I'm not sure if I'm missing something myself, but `sudo pip3 install adafruit-circuitpython-ssd1306` should just work, I'd expect.
I also followed the steps described in [Installing CircuitPython Libraries on Raspberry Pi](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi).
Thank you very much.

makermelissa, 2022-02-11:
Fixed via #552.

closed 2022-02-11 by makermelissa

meandthemachine, 2022-02-14:
@makermelissa Thanks a lot for keeping track!