adafruit/Adafruit_Blinka #551: Support - NanoPi R2S

marcindotkot, 2022-02-11:
Please add support for NanoPi R2S:

Details: https://wiki.friendlyarm.com/wiki/index.php/NanoPi_R2S

GPIO PIN Layout:
Pin# | Name | Linux gpio | Pin# | Name | Linux gpio
1 | SYS_3.3V |   | 
2 | VDD_5V |  
3 | I2C0_SDA / GPIO2_D1 | 89 | 
4 | VDD_5V |  
5 | I2C0_SCL / GPIO2_D0 | 88 | 
6 | GND |  
7 | GPIO2_A2 / IR_RX | 66 | 
8 | UART1_TX / GPIO3_A4 | 100
9 | GND |   | 
10 | UART1_RX / GPIOG3_A6 | 102

ladyada, 2022-02-11:
we dont own one so anyone who does can please submit a PR thanks :)

makermelissa, 2022-02-11:
You can check out these guides for information on how to add it:
https://learn.adafruit.com/adding-a-single-board-computer-to-platformdetect-for-blinka
https://learn.adafruit.com/adding-a-single-board-computer-to-blinka