adafruit/Adafruit_Blinka #553: Raspberry Pi 4B 8GB - NotImplementedError

stefan-hinterhoelzl, 2022-02-12:
Hi there,

I am having some issues with adafruit blinka on my raspberry pi 4b (8GB version).

I installed everything as defined in the guides. However, when trying to import the board module a "NotImplemented" error is thrown. 

This is the exact error:

Traceback (most recent call last):
  File "/home/stefan/Documents/Scripts/blinkatest.py", line 1, in <module>
    import board
  File "/home/stefan/.local/lib/python3.9/site-packages/board.py", line 285, in <module>
    raise NotImplementedError(
NotImplementedError: Adafruit-PlatformDetect version 3.19.5 was unable to identify the board and/or microcontroller running the Linux platform. Please be sure you have the latest packages running: 'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

From the error message i reckon the pi 4b 8Gb is somehow not supported. This is hard to believe though, since the pi 4b isn't that new and listed as supported on the circuitpython webpage.

Looking forward to some help!

Regards

stefan-hinterhoelzl, 2022-02-12:
Alright, looked closer through the modules.
Seems like my rev code of the raspi 4B is not included in the boards.py in the platform detector.

Here the Output of the Hardware
stefan@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: d03115
Serial		: 100000003462d3b2
Model		: Raspberry Pi 4 Model B Rev 1.5

boards.py only includes d03114, not d03115.
I think this should fix the issue. I will try this tomorrow, and if so, put together a PR for it.

ladyada, 2022-02-12:
yep please submit a PR, if you can :)

stefan-hinterhoelzl, 2022-02-13:
I created the PR.
So this issue can be closed now :)

closed 2022-02-13 by stefan-hinterhoelzl