adafruit/Adafruit_Blinka #563: `import board` with MCP2221 hangs

dalbabur, 2022-03-09:
I'm on a raspberry Pi 4 armv7l with Bullseye . I followed the setup from [https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/](https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/)

 No hid_mcp2221 driver was found (`rmmod: ERROR: Module hid_mcp2221 is not currently loaded` ), but I still blacklisted.
 I have set `BLINKA_MCP2221 = 1` and `BLINKA_MCP2221_RESET_DELAY =  #20`

`hidapi` post-install checks run OK.

when calling `import board`, either python hangs or fails with `OSError: open failed`

Related to #535

caternuson, 2022-03-09:
Which specific OS image release? What release date and is it the Desktop or Lite version?

Why have you set `BLINKA_MCP2221_RESET_DELAY = #20`?

dalbabur, 2022-03-09:
Linux raspberrypi 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux. Desktop version

I originally did not change the RESET_DELAY, tried it because of #535

caternuson, 2022-03-09:
Can not recreate this issue. Please post in the forums with more details about your setup. We need to determine if it's a hardware related issue before creating an issue in this repo:
https://forums.adafruit.com/

Went through a new install of Raspberry Pi OS Desktop starting with same release as above and tested on a Pi 4.

It appears that the Pi does not have the native driver. So the need to remove and blacklist `hid_mcp2221` is not really needed. But following the guide should also not affect things. It just won't really do anything.
```console
[ 3516.880882] usb 1-1.3: new full-speed USB device number 3 using xhci_hcd
[ 3517.024490] usb 1-1.3: New USB device found, idVendor=04d8, idProduct=00dd, bcdDevice= 1.00
[ 3517.024512] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3517.024531] usb 1-1.3: Product: MCP2221 USB-I2C/UART Combo
[ 3517.024549] usb 1-1.3: Manufacturer: Microchip Technology Inc.
[ 3517.053233] hid-generic 0003:04D8:00DD.0001: hiddev96,hidraw0: USB HID v1.11 Device [Microchip Technology Inc. MCP2221 USB-I2C/UART Combo] on usb-0000:01:00.0-1.3/input2
pi@raspberrypi:~ $ lsmod | grep hid
pi@raspberrypi:~ $
```

Other than weird assertions related to some mutex clean up when exiting the one test, everything worked as expected:
```console
pi@raspberrypi:~ $ pip3 install hidapi
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting hidapi
  Downloading https://www.piwheels.org/simple/hidapi/hidapi-0.11.0.post2-cp39-cp39-linux_armv7l.whl (348 kB)
     |████████████████████████████████| 348 kB 2.1 kB/s 
Requirement already satisfied: setuptools>=19.0 in /usr/lib/python3/dist-packages (from hidapi) (52.0.0)
Installing collected packages: hidapi
Successfully installed hidapi-0.11.0.post2
pi@raspberrypi:~ $ pip3 install adafruit-blinka
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting adafruit-blinka
  Downloading https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-7.1.1-py3-none-any.whl (216 kB)
     |████████████████████████████████| 216 kB 7.8 kB/s 
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from adafruit-blinka) (0.7.0)
Collecting pyftdi>=0.40.0
  Downloading https://www.piwheels.org/simple/pyftdi/pyftdi-0.53.3-py3-none-any.whl (141 kB)
     |████████████████████████████████| 141 kB 122 kB/s 
Collecting Adafruit-PureIO>=1.1.7
  Downloading https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.9-py3-none-any.whl (12 kB)
Collecting Adafruit-PlatformDetect>=3.13.0
  Downloading https://www.piwheels.org/simple/adafruit-platformdetect/Adafruit_PlatformDetect-3.21.0-py3-none-any.whl (18 kB)
Collecting sysv-ipc>=1.1.0
  Downloading https://www.piwheels.org/simple/sysv-ipc/sysv_ipc-1.1.0-cp39-cp39-linux_armv7l.whl (63 kB)
     |████████████████████████████████| 63 kB 2.5 kB/s 
Collecting rpi-ws281x>=4.0.0
  Downloading https://www.piwheels.org/simple/rpi-ws281x/rpi_ws281x-4.3.3-cp39-cp39-linux_armv7l.whl (114 kB)
     |████████████████████████████████| 114 kB 9.4 kB/s 
Requirement already satisfied: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi>=0.40.0->adafruit-blinka) (3.5b0)
Collecting pyusb!=1.2.0,>=1.0.0
  Downloading https://www.piwheels.org/simple/pyusb/pyusb-1.2.1-py3-none-any.whl (58 kB)
     |████████████████████████████████| 58 kB 139 kB/s 
Installing collected packages: pyusb, sysv-ipc, rpi-ws281x, pyftdi, Adafruit-PureIO, Adafruit-PlatformDetect, adafruit-blinka
Successfully installed Adafruit-PlatformDetect-3.21.0 Adafruit-PureIO-1.1.9 adafruit-blinka-7.1.1 pyftdi-0.53.3 pyusb-1.2.1 rpi-ws281x-4.3.3 sysv-ipc-1.1.0
pi@raspberrypi:~ $ python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> hid.enumerate()
[{'path': b'1-1.3:1.2', 'vendor_id': 1240, 'product_id': 221, 'serial_number': '', 'release_number': 256, 'manufacturer_string': 'Microchip Technology Inc.', 'product_string': 'MCP2221 USB-I2C/UART Combo', 'usage_page': 0, 'usage': 0, 'interface_number': 2}]
>>> exit()
pi@raspberrypi:~ $ python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> device = hid.device()
>>> device.open(0x04D8, 0x00DD)
>>> exit()
python3: ../../libusb/os/threads_posix.h:58: usbi_mutex_destroy: Assertion `pthread_mutex_destroy(mutex) == 0' failed.
python3: ../../libusb/os/threads_posix.h:46: usbi_mutex_lock: Assertion `pthread_mutex_lock(mutex) == 0' failed.
Aborted
pi@raspberrypi:~ $ export BLINKA_MCP2221=1
pi@raspberrypi:~ $ python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.environ["BLINKA_MCP2221"]
'1'
>>> import board
>>> dir(board)
['G0', 'G1', 'G2', 'G3', 'I2C', 'SCL', 'SDA', '__blinka__', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__repo__', '__spec__', '__version__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']
>>> 
```

closed 2022-03-09 by caternuson

dalbabur, 2022-03-09:
interesting. thanks for taking the time to go through all the testing