adafruit/Adafruit_Blinka #564: Add support for Khadas VIM3 Board.

BytEvil, 2022-03-14:
Hi, friends! Please add support for Khadas VIM 3 Board (Amlogic A311D chip).

/sys/kernel/debug/pinctrl/
`ff634400.bus:pinctrl@40-pinctrl-meson  ff800000.sys-ctrl:pinctrl@14-pinctrl-meson`

/sys/kernel/debug/pinctrl/ff634400.bus:pinctrl@40-pinctrl-meson/gpio-ranges
```
GPIO ranges handled:
0: periphs-banks GPIOS [427 - 512] PINS [0 - 85]
```

/sys/kernel/debug/pinctrl/ff634400.bus:pinctrl@40-pinctrl-meson/pins
```
registered pins: 85
pin 0 (GPIOZ_0) 0:periphs-banks  ff634400.bus:pinctrl@40
pin 1 (GPIOZ_1) 1:periphs-banks  ff634400.bus:pinctrl@40
pin 2 (GPIOZ_2) 2:periphs-banks  ff634400.bus:pinctrl@40
pin 3 (GPIOZ_3) 3:periphs-banks  ff634400.bus:pinctrl@40
pin 4 (GPIOZ_4) 4:periphs-banks  ff634400.bus:pinctrl@40
pin 5 (GPIOZ_5) 5:periphs-banks  ff634400.bus:pinctrl@40
pin 6 (GPIOZ_6) 6:periphs-banks  ff634400.bus:pinctrl@40
pin 7 (GPIOZ_7) 7:periphs-banks  ff634400.bus:pinctrl@40
pin 8 (GPIOZ_8) 8:periphs-banks  ff634400.bus:pinctrl@40
pin 9 (GPIOZ_9) 9:periphs-banks  ff634400.bus:pinctrl@40
pin 10 (GPIOZ_10) 10:periphs-banks  ff634400.bus:pinctrl@40
pin 11 (GPIOZ_11) 11:periphs-banks  ff634400.bus:pinctrl@40
pin 12 (GPIOZ_12) 12:periphs-banks  ff634400.bus:pinctrl@40
pin 13 (GPIOZ_13) 13:periphs-banks  ff634400.bus:pinctrl@40
pin 14 (GPIOZ_14) 14:periphs-banks  ff634400.bus:pinctrl@40
pin 15 (GPIOZ_15) 15:periphs-banks  ff634400.bus:pinctrl@40
pin 16 (GPIOH_0) 16:periphs-banks  ff634400.bus:pinctrl@40
pin 17 (GPIOH_1) 17:periphs-banks  ff634400.bus:pinctrl@40
pin 18 (GPIOH_2) 18:periphs-banks  ff634400.bus:pinctrl@40
pin 19 (GPIOH_3) 19:periphs-banks  ff634400.bus:pinctrl@40
pin 20 (GPIOH_4) 20:periphs-banks  ff634400.bus:pinctrl@40
pin 21 (GPIOH_5) 21:periphs-banks  ff634400.bus:pinctrl@40
pin 22 (GPIOH_6) 22:periphs-banks  ff634400.bus:pinctrl@40
pin 23 (GPIOH_7) 23:periphs-banks  ff634400.bus:pinctrl@40
pin 24 (GPIOH_8) 24:periphs-banks  ff634400.bus:pinctrl@40
pin 25 (BOOT_0) 25:periphs-banks  ff634400.bus:pinctrl@40
pin 26 (BOOT_1) 26:periphs-banks  ff634400.bus:pinctrl@40
pin 27 (BOOT_2) 27:periphs-banks  ff634400.bus:pinctrl@40
pin 28 (BOOT_3) 28:periphs-banks  ff634400.bus:pinctrl@40
pin 29 (BOOT_4) 29:periphs-banks  ff634400.bus:pinctrl@40
pin 30 (BOOT_5) 30:periphs-banks  ff634400.bus:pinctrl@40
pin 31 (BOOT_6) 31:periphs-banks  ff634400.bus:pinctrl@40
pin 32 (BOOT_7) 32:periphs-banks  ff634400.bus:pinctrl@40
pin 33 (BOOT_8) 33:periphs-banks  ff634400.bus:pinctrl@40
pin 34 (BOOT_9) 34:periphs-banks  ff634400.bus:pinctrl@40
pin 35 (BOOT_10) 35:periphs-banks  ff634400.bus:pinctrl@40
pin 36 (BOOT_11) 36:periphs-banks  ff634400.bus:pinctrl@40
pin 37 (BOOT_12) 37:periphs-banks  ff634400.bus:pinctrl@40
pin 38 (BOOT_13) 38:periphs-banks  ff634400.bus:pinctrl@40
pin 39 (BOOT_14) 39:periphs-banks  ff634400.bus:pinctrl@40
pin 40 (BOOT_15) 40:periphs-banks  ff634400.bus:pinctrl@40
pin 41 (GPIOC_0) 41:periphs-banks  ff634400.bus:pinctrl@40
pin 42 (GPIOC_1) 42:periphs-banks  ff634400.bus:pinctrl@40
pin 43 (GPIOC_2) 43:periphs-banks  ff634400.bus:pinctrl@40
pin 44 (GPIOC_3) 44:periphs-banks  ff634400.bus:pinctrl@40
pin 45 (GPIOC_4) 45:periphs-banks  ff634400.bus:pinctrl@40
pin 46 (GPIOC_5) 46:periphs-banks  ff634400.bus:pinctrl@40
pin 47 (GPIOC_6) 47:periphs-banks  ff634400.bus:pinctrl@40
pin 48 (GPIOC_7) 48:periphs-banks  ff634400.bus:pinctrl@40
pin 49 (GPIOA_0) 49:periphs-banks  ff634400.bus:pinctrl@40
pin 50 (GPIOA_1) 50:periphs-banks  ff634400.bus:pinctrl@40
pin 51 (GPIOA_2) 51:periphs-banks  ff634400.bus:pinctrl@40
pin 52 (GPIOA_3) 52:periphs-banks  ff634400.bus:pinctrl@40
pin 53 (GPIOA_4) 53:periphs-banks  ff634400.bus:pinctrl@40
pin 54 (GPIOA_5) 54:periphs-banks  ff634400.bus:pinctrl@40
pin 55 (GPIOA_6) 55:periphs-banks  ff634400.bus:pinctrl@40
pin 56 (GPIOA_7) 56:periphs-banks  ff634400.bus:pinctrl@40
pin 57 (GPIOA_8) 57:periphs-banks  ff634400.bus:pinctrl@40
pin 58 (GPIOA_9) 58:periphs-banks  ff634400.bus:pinctrl@40
pin 59 (GPIOA_10) 59:periphs-banks  ff634400.bus:pinctrl@40
pin 60 (GPIOA_11) 60:periphs-banks  ff634400.bus:pinctrl@40
pin 61 (GPIOA_12) 61:periphs-banks  ff634400.bus:pinctrl@40
pin 62 (GPIOA_13) 62:periphs-banks  ff634400.bus:pinctrl@40
pin 63 (GPIOA_14) 63:periphs-banks  ff634400.bus:pinctrl@40
pin 64 (GPIOA_15) 64:periphs-banks  ff634400.bus:pinctrl@40
pin 65 (GPIOX_0) 65:periphs-banks  ff634400.bus:pinctrl@40
pin 66 (GPIOX_1) 66:periphs-banks  ff634400.bus:pinctrl@40
pin 67 (GPIOX_2) 67:periphs-banks  ff634400.bus:pinctrl@40
pin 68 (GPIOX_3) 68:periphs-banks  ff634400.bus:pinctrl@40
pin 69 (GPIOX_4) 69:periphs-banks  ff634400.bus:pinctrl@40
pin 70 (GPIOX_5) 70:periphs-banks  ff634400.bus:pinctrl@40
pin 71 (GPIOX_6) 71:periphs-banks  ff634400.bus:pinctrl@40
pin 72 (GPIOX_7) 72:periphs-banks  ff634400.bus:pinctrl@40
pin 73 (GPIOX_8) 73:periphs-banks  ff634400.bus:pinctrl@40
pin 74 (GPIOX_9) 74:periphs-banks  ff634400.bus:pinctrl@40
pin 75 (GPIOX_10) 75:periphs-banks  ff634400.bus:pinctrl@40
pin 76 (GPIOX_11) 76:periphs-banks  ff634400.bus:pinctrl@40
pin 77 (GPIOX_12) 77:periphs-banks  ff634400.bus:pinctrl@40
pin 78 (GPIOX_13) 78:periphs-banks  ff634400.bus:pinctrl@40
pin 79 (GPIOX_14) 79:periphs-banks  ff634400.bus:pinctrl@40
pin 80 (GPIOX_15) 80:periphs-banks  ff634400.bus:pinctrl@40
pin 81 (GPIOX_16) 81:periphs-banks  ff634400.bus:pinctrl@40
pin 82 (GPIOX_17) 82:periphs-banks  ff634400.bus:pinctrl@40
pin 83 (GPIOX_18) 83:periphs-banks  ff634400.bus:pinctrl@40
pin 84 (GPIOX_19) 84:periphs-banks  ff634400.bus:pinctrl@40

```
/sys/kernel/debug/pinctrl/ff800000.sys-ctrl:pinctrl@14-pinctrl-meson/gpio-ranges
```
GPIO ranges handled:
0: aobus-banks GPIOS [412 - 426] PINS [0 - 14]
```

/sys/kernel/debug/pinctrl/ff800000.sys-ctrl:pinctrl@14-pinctrl-meson/pins
```
registered pins: 15
pin 0 (GPIOAO_0) 0:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 1 (GPIOAO_1) 1:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 2 (GPIOAO_2) 2:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 3 (GPIOAO_3) 3:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 4 (GPIOAO_4) 4:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 5 (GPIOAO_5) 5:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 6 (GPIOAO_6) 6:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 7 (GPIOAO_7) 7:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 8 (GPIOAO_8) 8:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 9 (GPIOAO_9) 9:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 10 (GPIOAO_10) 10:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 11 (GPIOAO_11) 11:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 12 (GPIOE_0) 12:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 13 (GPIOE_1) 13:aobus-banks  ff800000.sys-ctrl:pinctrl@14
pin 14 (GPIOE_2) 14:aobus-banks  ff800000.sys-ctrl:pinctrl@14
```

40-pin header layout:

```
 1 | 21 :			                    5V | GND
 2 | 22 :			                    5V | GPIOA_15
 3 | 23 :		                  USB_DM | GPIOA_14
 4 | 24 :                                USB_DP | GND
 5 | 25 :	                                GND | GPIOAO_2
 6 | 26 :	      	                   MCU3.3 | GPIOAO_3
 7 | 27 :			     MCU_NRST | 3.3V
 8 | 28 :		 	    MCU_SWIM | GND
 9 | 29 :	                               GND | GPIOA_1
10 | 30 :	                       ADC0	|  GPIOA_0
11 | 31 :		                 1.8V	| GPIOA_3
12 | 32 :		               ADC1	| GPIOA_2
13 | 33 :	                    GPIOAO_10 | GPIOA_4
14 | 34 :		                       GND	| GND
15 | 35 :  (GPIOH_6)UART_RX_AO_B | PWM-F(GPIOH_5)
16 | 36 :  (GPIOH_7)UART_TX_AO_B | RTC_CLK
17 | 37 :				GND	| GPIOH_4
18 | 38 :	 (GPIOAO_1)Linux_RX	| MCUPA_1
19 | 39 :	      (GPIOAO_0)Linux_TX | GPIOZ_15
20 | 40 :				   3.3V	| GND
```

makermelissa, 2022-03-15:
Hi, we don't have one of these boards. Would you like to add it? It looks like you've already done the hard part which is figuring out which pins connect to which GPIOs. Here's some guides to help you:
https://learn.adafruit.com/adding-a-single-board-computer-to-platformdetect-for-blinka
https://learn.adafruit.com/adding-a-single-board-computer-to-blinka

makermelissa, 2022-03-30:
Fixed via #566.