adafruit/Adafruit_Blinka #565: Missing support for RADXA Rock3 Model A

thecalle, 2022-03-15:
Chip RK3568 - 40 PIN Header V1.3/V1.31

[https://wiki.radxa.com/Rock3/hardware/rock3a-gpio](url)

Board is not detected.

makermelissa, 2022-03-15:
Hi, we don't have one of these boards. Would you like to add it. Here's some guides to help you:
https://learn.adafruit.com/adding-a-single-board-computer-to-platformdetect-for-blinka
https://learn.adafruit.com/adding-a-single-board-computer-to-blinka

makermelissa, 2022-03-15:
It looks like we have the RockPi 4 (https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/board/radxa/rockpi4.py). Maybe you could use that as a starting point.

thecalle, 2022-03-15:
Ok. I will proceed using RockPI4 as a reference

makermelissa, 2022-03-31:
Fixed via #566

closed 2022-03-31 by makermelissa