adafruit/Adafruit_Blinka #567: Example Code on Code Page Not working ...

sealyons, 2022-03-19:
I tried to run the Example code on the Code Page and I get a traceback saying the module 'testing' can't be found.  Does this code still supposed to be working?  I'm trying to run on a Raspberry Pi 400.

from testing import test_module_name
test_module_name("testing.universal.digitalio")

dhalbert, 2022-03-19:
`testing` is the module in this directory: https://github.com/adafruit/Adafruit_Blinka/tree/main/test/src/testing

sealyons, 2022-03-19:
Hi Dan,This really doesn't help me. My issue is I get a teaceback when I try to access the testing module from my Pi400. Everything else appears to be accessible. If the testing module exists in the latest release, why am I getting a traceback? Appreciate any additional guidance. Thanks, Chris Lyons

makermelissa, 2022-06-09:
First of all, the tests were originally written when Blinka only worked on MicroPython. I don't think testing has been used or updated in a long time (like 3-4 years). Since testing is because it is outside of the root src folder (https://github.com/adafruit/Adafruit_Blinka/tree/main/src), the way to run it is to first make sure you have cloned the Blinka repo. Then go into the test/src folder and run it. It should start running the test suite, but due to not being updated in years, it will likely fail.

closed 2022-06-09 by makermelissa