adafruit/Adafruit_Blinka #569: GPIO on PI pico with circuitpython not working.

niiccoo2, 2022-03-20:
Code is:
```
import time
import board
import digitalio
import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode

# Set up a keyboard device.
kbd = Keyboard(usb_hid.devices)

s1 = digitalio.DigitalInOut(board.GP16)
s1.switch_to_input(pull=digitalio.Pull.DOWN)

s2 = digitalio.DigitalInOut(board.GP21)
s2.switch_to_input(pull=digitalio.Pull.DOWN)

last1 = s1.value

last2 = s2.value

while True:
    print(s1.value)
    if(s1.vgalue != last1):
        if(s1.value):
            print("Landing gear down!")
            kbd.send(Keycode.G)
        else:
            print("Landing gear up")
            kbd.send(Keycode.G)
    if(s2.vgalue != last2):
        if(s1.value):
            print("Brakes On!")
            kbd.send(Keycode.B)
        else:
            print("Brakes Off!")
            kbd.send(Keycode.B)
    last1 = s1.value
    last2 = s2.value
    time.sleep(0.5)
```

Output is:
```

Adafruit CircuitPython 7.2.3 on 2022-03-16; Raspberry Pi Pico with rp2040
>>> %Run -c $EDITOR_CONTENT
False
Traceback (most recent call last):
  File "<stdin>", line 24, in <module>
AttributeError: 'DigitalInOut' object has no attribute 'vgalue'
>>> 
```

dhalbert, 2022-03-20:
You have a spelling error. Change `vgalue` to `value`.

By the way, this is not a Blinka issue.

closed 2022-03-20 by ladyada

niiccoo2, 2022-03-20:
OMG sorry was up late;)