adafruit/Adafruit_Blinka #570: ImportError: no module named 'typing'

Michel-HEG, 2022-03-23:
with my raspberrypi pico i have this error, is there a way to have this module??

dhalbert, 2022-03-23:
This module is not meant for use on the Pi Pico. What code is throwing this error? In our libraries, we usually do something like:
```py
try:
    from typing import ...   # some things
except ImportError:
    pass
```

Michel-HEG, 2022-03-23:
i'm trying to use HID and that module imports that, i have this error:

Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "/lib/usb_hid.py", line 14, in <module>
ImportError: no module named 'typing'
>>>

Michel-HEG, 2022-03-23:
yes i'm doing that, i will check your link thanks

Michel-HEG, 2022-03-23:
i used try catch: but it doesn't work, i have the same error with evry other modules below now:
`>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "/lib/usb_hid.py", line 19, in <module>
ImportError: no module named 'pathlib'
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
ImportError: no module named 'atexit'
>>> `

DanielTomlinsonSage, 2022-03-28:
usb_hid.py no longer exists! No alternative given, same general issue.

makermelissa, 2022-03-28:
> usb_hid.py no longer exists! No alternative given, same general issue.

Do you mean this file? https://github.com/adafruit/Adafruit_Blinka/blob/main/src/usb_hid.py

makermelissa, 2022-03-31:
`usb_hid.py` was recently added by a community member to make use of the usb_gadget driver, which is only on linux. I'm not sure what it would take to make use of something similar within MicroPython.

makermelissa, 2022-06-02:
Fixed via https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/230 and #586.

closed 2022-06-02 by makermelissa