adafruit/Adafruit_Blinka #571: gpiod.Chip should not use uppercase

Neradoc, 2022-03-25:
Somebody reported this on discord:
```
# uname -r
5.10.102-meson64
# cat /etc/debian_version
bullseye/sid
# python
Python 3.8.10 (default, Nov 26 2021, 20:14:08)
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/board.py", line 158, in <module>
    from adafruit_blinka.board.hardkernel.odroidc2 import *
  File "/usr/local/lib/python3.8/dist-packages/adafruit_blinka/board/hardkernel/odroidc2.py", line 6, in <module>
    from adafruit_blinka.microcontroller.amlogic.s905 import pin
  File "/usr/local/lib/python3.8/dist-packages/adafruit_blinka/microcontroller/amlogic/s905/pin.py", line 7, in <module>
    GPIO122 = Pin((0, 0))
  File "/usr/local/lib/python3.8/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 34, in __init__
    self._chip = gpiod.Chip(str(pin_id[0]), gpiod.Chip.OPEN_BY_NUMBER)
AttributeError: module 'gpiod' has no attribute 'Chip'
```
Looking at the gpiod repo it does look like it should be `gpiod.chip` in lowercase ?
Or could there be a different version of the python gpiod module that conflicts ?

makermelissa, 2022-03-25:
If I recall correctly, older versions of libgpiod used `Chip` instead of `chip`, but maybe something changed again? I can look into this probably next week. I have the Odroid C2 on hand.

makermelissa, 2022-03-25:
If it is a version thing, it might be worth adding a version check of the library to Blinka since this sin't the first person to fun into this.

makermelissa, 2022-03-29:
if somebody is having this error, please run `pip3 install gpiod` or `sudo pip3 install gpiod`. I'm currently working on updating the guide. Apparently installing via `sudo apt-get` installs an older version.

dhalbert, 2022-03-29:
You can just `hasattr("gpiod", "Chip")` and then monkey-patch it: `gpiod.chip = gpiod.Chip` or something like that.

makermelissa, 2022-03-29:
> You can just `hasattr("gpiod", "Chip")` and then monkey-patch it: `gpiod.chip = gpiod.Chip` or something like that.

I started doing that, but realized there's other differences like `gpiod.LINE_REQ_DIR_IN` vs `gpiod.line.DIRECTION_INPUT`. Anyway, there's like only 4 places, so I'm just using hasattr in each to begin with.

makermelissa, 2022-03-29:
I think I could add #425 at the same time since this is probably only on the newer version.

PR #574 (2022-03-29): Allow older and newer versions of libgpiod to work, Allow Pullups
Fixes #571. Fixes #425. Tested on Odroid C2 with older and newer libgpiod.
closed 2022-03-29 by dhalbert