adafruit/Adafruit_Blinka #573: MC2221A import board RuntimeError: already open

nickGermi, 2022-03-29:
Hello,

When running following python code
`import os; os.environ['BLINKA_MCP2221'] = '1'
import os; os.environ['BLINKA_MCP2221_RESET_DELAY'] = '0.5'
import board`

I get following error:

`Traceback (most recent call last):
  File "c:\Users\Nick\Desktop\mc2221a.py", line 8, in <module>
    import board
  File "C:\Python39\lib\site-packages\board.py", line 182, in <module>
    from adafruit_blinka.board.microchip_mcp2221 import *
  File "C:\Python39\lib\site-packages\adafruit_blinka\board\microchip_mcp2221.py", line 5, in <module>
    from adafruit_blinka.microcontroller.mcp2221 import pin
  File "C:\Python39\lib\site-packages\adafruit_blinka\microcontroller\mcp2221\pin.py", line 5, in <module>
    from .mcp2221 import mcp2221
  File "C:\Python39\lib\site-packages\adafruit_blinka\microcontroller\mcp2221\mcp2221.py", line 389, in <module>
    mcp2221 = MCP2221()
  File "C:\Python39\lib\site-packages\adafruit_blinka\microcontroller\mcp2221\mcp2221.py", line 58, in __init__
    self._reset()
  File "C:\Python39\lib\site-packages\adafruit_blinka\microcontroller\mcp2221\mcp2221.py", line 128, in _reset
    self._hid.open(MCP2221.VID, MCP2221.PID)
  File "hid.pyx", line 111, in hid.device.open
RuntimeError: already open`

If I just set environment variable and open a Python shell and just type "import board" I get same exact error as above.

If I try to use PyMCP2221A module to communicate with my MCP221A device instead, it seems to be working fine. "MCP2221 I2C/SMBus Terminal" utility also seems to be working fine (I can read data via I2C protocol)

I'm using latest version of adafruit-blinka (7.1.1) & hidapi (0.11.2). I was using following guide to set everything up:
https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/windows

nickGermi, 2022-03-30:
So after commenting out following lines:

File: `lib\site-packages\adafruit_blinka\microcontroller\mcp2221\mcp2221.py`
```
61        # if MCP2221_RESET_DELAY >= 0:
62        #     self._reset()
```

`import board` works without generating any error. However, `board(dir)` produces `TypeError: 'module' object is not callable`
But, my code relevant to my sensor is now working fine

```
import os
os.environ['BLINKA_MCP2221'] = '1'
import board
import adafruit_tsl2591

i2c = board.I2C()
sensor = adafruit_tsl2591.TSL2591(i2c)
while True:
    lux = sensor.lux
    print("Total light: {0}lux".format(lux))
```

Produces:
> Total light: 108.638976lux
> Total light: 108.638976lux
> Total light: 108.802176lux
> Total light: 108.802176lux
> Total light: 108.802176lux
> Total light: 108.802176lux
> Total light: 108.53452800000001lux
> Total light: 108.53452800000001lux

Based on this finding, I did some more testing and reverted back the changes I made to line 61 & 62 followed by setting following value in my script:
`os.environ['BLINKA_MCP2221_RESET_DELAY'] = '-1'`

With above value, my code works fine without having to modify the file `lib\site-packages\adafruit_blinka\microcontroller\mcp2221\mcp2221.py`. That said, I still get `TypeError: 'module' object is not callable` when trying `board(dir)`

Leaving this post open for you to decide what you want to do, also happy to provide more test result if needed

caternuson, 2022-03-30:
Looks like a weird behavior related to the resetting. Setting `BLINKA_MCP2221_RESET_DELAY=-1` is a good way to prevent the reset from happening without needing to change the library source code. Using a larger non-zero value for `BLINKA_MCP2221_RESET_DELAY` also does not seem to help. Might be some new Windows 11 related behavior?

> However, board(dir) produces TypeError: 'module' object is not callable

This is a basic Python syntax error. The correct command is:
```python
dir(board)
```
`dir` is the command, `board` is the parameter to the command

caternuson, 2022-03-30:
Adding potentially related forum post for tie in:
https://forums.adafruit.com/viewtopic.php?f=60&t=189781

AaronKehl, 2022-04-06:
Just FYSA, same situation happening on my machine. Ubuntu 20.04.  `export BLINKA_MCP2221_RESET_DELAY=-1` worked, as mentioned, for me.

Neradoc, 2022-04-13:
I see the same issue on Mac, 10.15.7, and it used to work relatively recently.
Reverting hidapi to the previous release fixed it.
```
pip install hidapi==0.11.0
```

I made the following change to `MCP2221._reset` instead.
```diff
@@ -137,6 +137,7 @@ class MCP2221:
         start = time.monotonic()
         while time.monotonic() - start < 5:
             try:
+                self._hid.close()
                 self._hid.open(MCP2221.VID, MCP2221.PID)
             except OSError:
                 # try again
```

Neradoc, 2022-04-13:
Looks like it comes from this, 5 months ago. (That repo is the source [for pypi's hidapi](https://pypi.org/project/hidapi/)).
https://github.com/trezor/cython-hidapi/commit/2cc4c3acbd7affd0e23c5f4db0753a8b4622d954

>  Raise RuntimeError if opening an already open device
> This is necessary so open devices don't leak, later crashing hid_exit().

caternuson, 2022-04-13:
@Neradoc Good catch. Looks like I had a slightly older version (0.10.1) of hidapi installed. Just upgraded to 0.11.2 and can now recreate the issue. Seems like maybe something isn't getting cleaned up correctly with certain sequences, like software resetting.

Reverting to an older version of hidapi should work as another work around for now:
```
pip install hidapi==0.10.1
```

caternuson, 2022-04-25:
Please try updating to the 7.2.2 release of Blinka:
https://github.com/adafruit/Adafruit_Blinka/releases/tag/7.2.2

nickGermi, 2022-04-28:
I can confirm upgrading from installed `adafruit-blinka-7.1.2` to `adafruit-blinka-7.2.2` fixed the issue, after the upgrade setting `BLINKA_MCP2221_RESET_DELAY` to `-1`, `0.5` or not setting any value at all works fine.

closed 2022-04-28 by nickGermi