adafruit/Adafruit_Blinka #575: rp2040: Missing I2C ports

Gadgetoid, 2022-04-13:
The `i2cPorts` list for rp2040 is incomplete.

See: https://github.com/adafruit/Adafruit_Blinka/blob/c4d26d1a2f3d037a20aa5ae55936baa265258edb/src/adafruit_blinka/microcontroller/rp2040/pin.py#L91-L124

Versus available combinations detailed on: https://pico.pinout.xyz/

Is this intentional? It looks like the ports use permutations which means a complete list would be rather a lot longer.

I had a user attempting to use I2C on GP21 / GP20 which is a valid combination for i2c0 and this failed with a very long-winded error:

```
ValueError: No Hardware I2C on (scl,sda)=(board.GP21, board.GP20)
Valid I2C ports: ((0, board.GP1, board.GP0), (0, board.GP1, board.GP4), (0, board.GP1, board.GP8), (0, board.GP1, board.GP12), (0, board.GP5, board.GP0), (0, board.GP5, board.GP4), (0, board.GP5, board.GP8), (0, board.GP5, board.GP12), (0, board.GP9, board.GP0), (0, board.GP9, board.GP4), (0, board.GP9, board.GP8), (0, board.GP9, board.GP12), (0, board.GP13, board.GP0), (0, board.GP13, board.GP4), (0, board.GP13, board.GP8), (0, board.GP13, board.GP12), (1, board.GP3, board.GP2), (1, board.GP3, board.GP6), (1, board.GP3, board.GP10), (1, board.GP3, board.GP14), (1, board.GP7, board.GP2), (1, board.GP7, board.GP6), (1, board.GP7, board.GP10), (1, board.GP7, board.GP14), (1, board.GP11, board.GP2), (1, board.GP11, board.GP6), (1, board.GP11, board.GP10), (1, board.GP11, board.GP14), (1, board.GP15, board.GP2), (1, board.GP15, board.GP6), (1, board.GP15, board.GP10), (1, board.GP15, board.GP14))
```

An exhaustive list of port permutations might get unwieldy. Would it be appropriate to - in the case of the Pico/RP2040 - replace this with the logic used in the SDK, MicroPython and CircuitPython C/C++ code.

IE: since i2c is available on (almost) all pins with a common pattern the `SCL % 2 == 1 and SDA % 2 == 0` check should suffice.

This check happens in CircuitPython here: https://github.com/adafruit/circuitpython/blob/7be66a57332fe63f6d2759663bf309b59162633d/ports/raspberrypi/common-hal/busio/I2C.c#L62-L69

makermelissa, 2022-04-13:
I don't think the logic that the other SDKs use would work with Blinka. Instead, some simple Python Code (perhaps using itertools) to generate the permutations would probably work splendidly.

Gadgetoid, 2022-04-13:
Using the power of being incredibly lazy I arrived at:

```python
for a in range(29):
    for b in range(29):
        inst_a = (a >> 1) % 2
        inst_b = (b >> 1) % 2
        if a % 2 == 1 and b % 2 == 0 and  inst_a == inst_b:
            print(f"({inst_a}, GP{a}, GP{b}),")
```

Which produces:

```python
(0, GP1, GP0),
(0, GP1, GP4),
(0, GP1, GP8),
(0, GP1, GP12),
(0, GP1, GP16),
(0, GP1, GP20),
(0, GP1, GP24),
(0, GP1, GP28),
(1, GP3, GP2),
(1, GP3, GP6),
(1, GP3, GP10),
(1, GP3, GP14),
(1, GP3, GP18),
(1, GP3, GP22),
(1, GP3, GP26),
(0, GP5, GP0),
(0, GP5, GP4),
(0, GP5, GP8),
(0, GP5, GP12),
(0, GP5, GP16),
(0, GP5, GP20),
(0, GP5, GP24),
(0, GP5, GP28),
(1, GP7, GP2),
(1, GP7, GP6),
(1, GP7, GP10),
(1, GP7, GP14),
(1, GP7, GP18),
(1, GP7, GP22),
(1, GP7, GP26),
(0, GP9, GP0),
(0, GP9, GP4),
(0, GP9, GP8),
(0, GP9, GP12),
(0, GP9, GP16),
(0, GP9, GP20),
(0, GP9, GP24),
(0, GP9, GP28),
(1, GP11, GP2),
(1, GP11, GP6),
(1, GP11, GP10),
(1, GP11, GP14),
(1, GP11, GP18),
(1, GP11, GP22),
(1, GP11, GP26),
(0, GP13, GP0),
(0, GP13, GP4),
(0, GP13, GP8),
(0, GP13, GP12),
(0, GP13, GP16),
(0, GP13, GP20),
(0, GP13, GP24),
(0, GP13, GP28),
(1, GP15, GP2),
(1, GP15, GP6),
(1, GP15, GP10),
(1, GP15, GP14),
(1, GP15, GP18),
(1, GP15, GP22),
(1, GP15, GP26),
(0, GP17, GP0),
(0, GP17, GP4),
(0, GP17, GP8),
(0, GP17, GP12),
(0, GP17, GP16),
(0, GP17, GP20),
(0, GP17, GP24),
(0, GP17, GP28),
(1, GP19, GP2),
(1, GP19, GP6),
(1, GP19, GP10),
(1, GP19, GP14),
(1, GP19, GP18),
(1, GP19, GP22),
(1, GP19, GP26),
(0, GP21, GP0),
(0, GP21, GP4),
(0, GP21, GP8),
(0, GP21, GP12),
(0, GP21, GP16),
(0, GP21, GP20),
(0, GP21, GP24),
(0, GP21, GP28),
(1, GP23, GP2),
(1, GP23, GP6),
(1, GP23, GP10),
(1, GP23, GP14),
(1, GP23, GP18),
(1, GP23, GP22),
(1, GP23, GP26),
(0, GP25, GP0),
(0, GP25, GP4),
(0, GP25, GP8),
(0, GP25, GP12),
(0, GP25, GP16),
(0, GP25, GP20),
(0, GP25, GP24),
(0, GP25, GP28),
(1, GP27, GP2),
(1, GP27, GP6),
(1, GP27, GP10),
(1, GP27, GP14),
(1, GP27, GP18),
(1, GP27, GP22),
(1, GP27, GP26)
```

This looks sorta right and isn't quite as bad as I'd envisioned.

makermelissa, 2022-04-13:
Looks good, but can we put the for loops in the `pin.py` file so it's generated on the fly? I fairly certain it's possible, but the trick is accessing the object properties with something like `getattr()`.

makermelissa, 2022-04-13:
In this case, it might actually be the `locals()` function.

Gadgetoid, 2022-04-13:
Ah, I see where you're going with this.

In the interest of heading off an inevitable headache, would it not be wise to hard-code them? This is a very RP2040 specific oddity that's not likely to change.

makermelissa, 2022-04-13:
Good point. Sure, we can hard code them.

dhalbert, 2022-04-13:
> Looks good, but can we put the for loops in the `pin.py` file so it's generated on the fly? I fairly certain it's possible, but the trick is accessing the object properties with something like `getattr()`.

Sure you can, it's just Python code that gets executed on import. I don't see why you need `getattr()`. The current list is already generated at run-time (tuples would be built at run-time), or the compiler might be optimizing the creation, but that's not inherent in Python.

dhalbert, 2022-04-13:
Ah, OK, sorry, you need `getattr()` for the pin names. But you can do that in the loops. Or maybe there should be a table of pin number -> pin name.

makermelissa, 2022-04-13:
Yeah, the table is a good idea. Then it makes things less complicated.

Gadgetoid, 2022-04-13:
Worth running up to 29, which adds this extra block? It's not relevant on Pico boards, but may be on RP2040-based alternatives.

```python
(0, GP29, GP0),
(0, GP29, GP4),
(0, GP29, GP8),
(0, GP29, GP12),
(0, GP29, GP16),
(0, GP29, GP20),
(0, GP29, GP24),
(0, GP29, GP28),
```

dhalbert, 2022-04-13:
Is there an alternate way to do this, to write this as a validator function instead of listing all the combos? That's how we do it in CircuitPython: we validate the combos in code. It might need some refactoring.

Gadgetoid, 2022-04-14:
@dhalbert this was roughly my initial proposal. But I think the case against complexity is that this is presently a very uniquely Pico/RP2040 problem. A quick glance over the other board definitions suggests most have no more than two i2cPorts so this explicit list approach works well.

It looks like any sort of validator:
* would require special case handling for Pico
* would obfuscate the available i2cPorts

In both cases this increases code complexity and debugging complexity to save a few lines.

I'd probably think about validators when the second or third board with dozens of valid i2c pin permutations crops up!

Of course I am biased by a vested interest in my PR being clean and uncomplicated :laughing: but I think these assertions are reasonable.

makermelissa, 2022-04-14:
https://github.com/adafruit/Adafruit_Blinka/blob/main/src/busio.py#L113-L119 is essentially where the data is used, so as long as it works with that, I'm happy with whichever solution you end up going with. :)

makermelissa, 2022-06-02:
Closing. I'm fairly certain this was fixed by #576.

closed 2022-06-02 by makermelissa