adafruit/Adafruit_Blinka #578: FT232H-connected EMC & OLED alternate in successful connection

rivimey, 2022-04-25:
I have an FT232H connected to an I2C mux, in turn connected (one per bus) to an OLED and 4x EMC2101 controllers. All parts from Adafruit except OLED. My controlling program uses the Adafruit python libs (nb: I'm also extending the emc2101 lib, but this issue is not dependent on that extension).

I have a program that connects fine to either the mux+oled or the mux+emc but not to both; the first connection the oled wins; Ctrl-C out and the EMC wins, then the OLED, etc. I don't have to change the python at all, just restart.

I don't understand what could be happening here... any ideas?

``` python
        from pyftdi.i2c import I2cNackError
        from pyftdi.usbtools import UsbToolsError

        try:
            import board
            import busio
        except UsbToolsError:
            return False
        except:
            return False

        import adafruit_tca9548a
        import adafruit_ssd1306
        from adafruit_emc2101 import emc2101_lut

        # Create I2C bus
        self.ft232h = busio.I2C(scl=board.SCL, sda=board.SDA, frequency=self.BUS_FREQ)
        self.pyftdi = self.ft232h._i2c._i2c.ftdi

        time.sleep(0.1)
        if _FANCTRL_DEBUG:
            print("Connected: ", self.pyftdi.is_connected)
            print("Configured: ", self.ft232h._i2c._i2c.configured)
            print("Frequency: ", self.ft232h._i2c._i2c.frequency)
            print("Latency: ", self.pyftdi.get_latency_timer())
            print("USB Path: ", self.pyftdi.usb_path)

            self.pyftdi.reset(usb_reset=False)
            # Make more responsive. Min 1, max 255, default 16
            self.pyftdi.set_latency_timer(12);
            print("Devices: ", self.pyftdi.list_devices())

        try:
            if _FANCTRL_DEBUG_SCAN:
                devs = self.ft232h.scan()
                print("I2C addresses found (ft):", [hex(devaddr) for devaddr in devs])

            time.sleep(0.1)
            # Create the TCA9548A object and give it the I2C bus
            try:
                self.tca = adafruit_tca9548a.TCA9548A(self.ft232h)
            except I2cNackError:
                if _FANCTRL_DEBUG:
                    print("Error: I2C Nack: tca")
            except ValueError:
                if _FANCTRL_DEBUG:
                    print("Error: No I2C TCA MUX device found")

            time.sleep(0.1)
            for bus in self.try_fan_channels:
                i2c_bus = self.tca[bus]
                try:
                    emc = emc2101_lut.EMC2101_LUT(i2c_bus)
                    emc.lut_enabled = True
                    self.emc.append(emc)
                    self.fan_channels.append(bus)
                except I2cNackError as e:
                    if _FANCTRL_DEBUG:
                        print("Error: I2C Nack: emc {}".format(bus))
                except AttributeError as e:
                    if _FANCTRL_DEBUG:
                        print("Error: No EMC I2C device on bus {}".format(bus))
                        print(e)
                except ValueError as e:
                    if _FANCTRL_DEBUG:
                        print("Error: No EMC I2C device on bus {}".format(bus))
                        print(e)
            time.sleep(0.1)
            bus = self.oled_channel
            i2c_bus = self.tca[bus]

            try:
                self.oled = adafruit_ssd1306.SSD1306_I2C(_OLED_W, _OLED_H, i2c_bus)

            except I2cNackError as ex:
                if _FANCTRL_DEBUG:
                    print("Error: I2C Nack: oled", bus, " ", ex)
                else:
                    print("Error: oled not configured on ", bus)
                return False

            except ValueError as ex:
                if _FANCTRL_DEBUG:
                    print("Error: No OLED driver found", bus, " ", ex)
                else:
                    print("Error: oled not configured on ", bus)
                return False

        # Make sure the bus is left unlocked, should something have
        # gone wrong.
        finally:
            try:
                self.ft232h.unlock()
            except:
                pass
            return True
```

This is the program not finding the EMC devices, but finding the OLED:

```
Connected:  True
Configured:  True
Frequency:  400000.0
Latency:  16
USB Path:  (1, 12, 0)
Error: No EMC I2C device on bus 0
No I2C device at address: 0x4c
Error: No EMC I2C device on bus 1
Cannot find a EMC2101
Error: No EMC I2C device on bus 2
Cannot find a EMC2101
Error: No EMC I2C device on bus 3
Cannot find a EMC2101
Error: No EMC I2C device on bus 4
Cannot find a EMC2101
Error: No EMC I2C device on bus 5
Cannot find a EMC2101
Error: No EMC I2C device on bus 6
Cannot find a EMC2101
Error: No EMC I2C device on bus 7
Cannot find a EMC2101
FT232H <busio.I2C object at 0x7f9906b88430>
TCA Mux <adafruit_tca9548a.TCA9548A object at 0x7f9907530310>
OLED <adafruit_ssd1306.SSD1306_I2C object at 0x7f9906b27a90>
 * Serving Flask app 'fanctrl' (lazy loading)
 * Serving Flask app 'fanctrl' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://127.0.0.1:5000 (Press CTRL+C to quit)
^CClosing...
```

and this is the reverse:

```
 * Running on http://127.0.0.1:5000 (Press CTRL+C to quit)
^CClosing...
[ruth@helva fanctrl.py]$ bin/python3 fanctrl.py
Connected:  True
Configured:  True
Frequency:  400000.0
Latency:  16
USB Path:  (1, 12, 0)
Error: No EMC I2C device on bus 0
No I2C device at address: 0x4c
Error: No EMC I2C device on bus 1
No I2C device at address: 0x4c
Error: No EMC I2C device on bus 2
No I2C device at address: 0x4c
Error: No EMC I2C device on bus 4
No I2C device at address: 0x4c
Error: No OLED driver found 0   No I2C device at address: 0x3c
FT232H <busio.I2C object at 0x7f63c19dc430>
TCA Mux <adafruit_tca9548a.TCA9548A object at 0x7f63c2385310>
OLED None
- EMC [<adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c197bc70>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c197be20>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c198a070>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c198a250>]
- EMC [<adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c197bc70>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c197be20>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c198a070>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c198a250>]
- EMC [<adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c197bc70>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c197be20>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c198a070>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c198a250>]
- EMC [<adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c197bc70>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c197be20>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c198a070>, <adafruit_emc2101.emc2101_lut.EMC2101_LUT object at 0x7f63c198a250>]
 * Serving Flask app 'fanctrl' (lazy loading)
```

caternuson, 2022-04-25:
Can you recreate this with simple test code? The snippet above looks like it's pulled out of some large class definition.

rivimey, 2022-04-25:
I'd really prefer not to... while you're right that there is a class definition, this is the whole of the setup code. 
I don't mind passing on the whole file (not long - about 400 lines in all, including ~180 lines of Flask REST stuff).

rivimey, 2022-04-25:
I have stripped out a lot of error checking and almost all of the Flask code, and set the emc mux numbers to just those that have a device, but because of no exception handling the oled now fails with a NAK error.

Strangely, no alternation happens now -- it's always the oled that fails!

```
import sys
import pprint
import time
import os
from flask import Flask, jsonify, request, abort, url_for, render_template
from functools import wraps

_FANCTRL_DEBUG = True
_FANCTRL_DEBUG_SCAN = False
_OLED_W = 128
_OLED_H = 32

###### ###### ###### ###### ###### ######

# Flask setup
app = Flask(__name__)

###### ###### ###### ###### ###### ######

class FanController:

    def __init__(self):
        # We are expecting a USB connection to the Adafruit FT232H i2c module
        os.environ["BLINKA_FT232"] = "1"
        os.environ["BLINKA_FORCEBOARD"] = "FTDI_FT232H"
        import board
        import busio

        # I2C at 1MHz
        self.BUS_FREQ = 400000
        # Channels to check for an EMC2101
        self.try_fan_channels = [3,5,6,7]
        # Channels on which an EMC2101 is found
        self.fan_channels = []
        self.oled_channel = 0

        self.emc = [] # Class objs for each fan
        self.oled = None
        self.tca = None
        self.ft232h = None
        self.pyftdi = None

    def setup(self):
        if not self.setup_bus():
            print("Error: FT232H/EMC2101 bus could not be configured.")
            return False

        for emc in self.emc:
            controller.setup_emc(emc)

        if self.oled != None:
            self.setup_oled()

        return True

    def setup_oled(self, emc):
        """Configure the OLED as needed"""
        self.oled.fill(0)
        self.oled.rect(2,1, _OLED_W-4, _OLED_H-2, 1)
        self.oled.fill_rect(5,3, _OLED_W-10, _OLED_H-6, 1)
        self.oled.show()

    def setup_emc(self, emc):
        """Configure the EMC2101 LUT and other params as needed"""
        emc.lut_enabled = True
        emc.lut[20] = 20
        emc.lut[30] = 35
        emc.lut[40] = 55
        emc.lut[50] = 80
        emc.lut[60] = 100
        emc.external_temp_low_limit = 10.0
        emc.external_temp_high_limit = 62.0
        return True

    def setup_bus(self):
        """Connect to the various devices - USB, Mux, Fans and OLED."""
        from pyftdi.i2c import I2cNackError
        from pyftdi.usbtools import UsbToolsError
        import board
        import busio
        import adafruit_tca9548a
        import adafruit_ssd1306
        from adafruit_emc2101 import emc2101_lut

        # Create I2C bus
        self.ft232h = busio.I2C(scl=board.SCL, sda=board.SDA, frequency=self.BUS_FREQ)
        self.pyftdi = self.ft232h._i2c._i2c.ftdi

        # Create the TCA9548A object and give it the I2C bus
        self.tca = adafruit_tca9548a.TCA9548A(self.ft232h)

        for bus in self.try_fan_channels:
            i2c_bus = self.tca[bus]
            emc = emc2101_lut.EMC2101_LUT(i2c_bus)
            emc.lut_enabled = True
            self.emc.append(emc)
            self.fan_channels.append(bus)

        bus = self.oled_channel
        i2c_bus = self.tca[bus]
        self.oled = adafruit_ssd1306.SSD1306_I2C(_OLED_W, _OLED_H, i2c_bus)

        return True

    def ft232(self):
        return self.ft232h

    def mux(self):
        return self.tca

    def emc(self):
        return self.emc

###### ###### ###### ###### ###### ######

@app.route('/', methods=['GET'])
def index():
    endpoints = {
        "/status": "General status",
    }
    return render_template("index.html", endpoints=endpoints)

###### ###### ###### ###### ###### ######

if __name__ == '__main__':
    #sys.tracebacklimit = 0
    controller = None

    try:
        controller = FanController()
    except Exception as e:
        print("Error: Fan controller USB Interface not found, is it plugged in?")
        print(e)
    else:
        if not controller.setup():
            print("Error: Unable to setup Fan controller")
            exit(1)
        app.run(host='127.0.0.1', use_reloader=False)

    finally:
        if controller != None:
            if controller.ft232h._i2c._i2c.configured:
                controller.ft232h._i2c._i2c.flush()
            print("Closing...")
            controller.ft232h._i2c._i2c.close()
            controller.ft232h._i2c._i2c.ftdi.close()

```

rivimey, 2022-04-25:
Ok, I think I know why oled fails all the time now - I thinned the set of mux values that an emc device is checked on, from 0..7 down to 3,5..7. Doing that with the original code also means the oled doesn't work, every time.

Could there be interference between the emc on addr 0x3c and the oled on addr 0x4c?

caternuson, 2022-04-25:
can you recreate the issue without involving flask?

ladyada, 2022-08-17:
closing the issue due to no response...