adafruit/Adafruit_Blinka #582: Slow loading time on RP2

LovelyA72, 2022-05-03:
Blinka libraries took a few seconds to load on a stock Pi Pico. Compared to instantaneous load on a CircuitPython device. Load time adds up really quickly when I am importing multiple libraries and it could easily take 10 sconds to load a program.

I am using the latest MicroPython for RP2 and the latest release of Blinka.

How to reproduce:
```py
import board
```

makermelissa, 2022-05-03:
Hi, Blinka is a library written in Python and it taking a little longer to run than native CircuitPython is expected. However, I will take a look when I get back from PyCon to make sure the delay seems reasonable or if it can be sped up at all.

makermelissa, 2022-05-06:
I think without heavy refactoring, the initial loading of Blinka is going to take a few second on a MicroController. The slowness is probably related to the board detection as subsequent imports are board are near instant. I'm wondering if some optimization can be done in PlatformDetect to check on the slower microcontrollers first.

makermelissa, 2022-05-06:
Unfortunately, I'm not seeing much to optimize here. I tried changing the order of checks in PlatformDetect, but it didn't make a difference. I'm going to close this as I'm not seeing much that can be done to get around the 2.5 second loading time.

closed 2022-05-06 by makermelissa