adafruit/Adafruit_Blinka #587: Beaglebone AI Support Issues

Andrew-xyz, 2022-05-27:
Hey all,

I saw that the Adafruit_Blinka library had added Beaglebone AI support, but haven't had any luck getting this running on my BBAI. I've initially tried to install via pip3, and to test, did the following:

`$ python3`
`>>> import board`

I received the following error message:

`ImportError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 83, in <module>
    from adafruit_blinka.board.beagleboard.beaglebone_ai import *
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/board/beagleboard/beaglebone_ai.py", line 5, in <module>
    from adafruit_blinka.microcontroller.dra74x import pin
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/dra74x/pin.py", line 5, in <module>
    from adafruit_blinka.microcontroller.generic_linux.libgpiod_pin import Pin
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 11, in <module>
    ) from ImportError
ImportError: libgpiod Python bindings not found, please install and try again! See https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/libgpiod.sh `

After running the installer script and attempting the same import, I now receive the following:

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 83, in <module>
    from adafruit_blinka.board.beagleboard.beaglebone_ai import *
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/board/beagleboard/beaglebone_ai.py", line 5, in <module>
    from adafruit_blinka.microcontroller.dra74x import pin
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/dra74x/pin.py", line 10, in <module>
    P8_3 = Pin((0, 24))  # GPIO1_24 - GPIO_24
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 35, in __init__
    self._chip = gpiod.Chip(str(pin_id[0]), gpiod.Chip.OPEN_BY_NUMBER)
AttributeError: type object 'gpiod.Chip' has no attribute 'OPEN_BY_NUMBER'`

My gpiod seems to be recognized, as I can do an `>>> import gpiod` in my python shell, but no luck getting the import to run. I haven't been able to find much information on using this library with the BBAI online, so maybe I'm missing something. Any help would be greatly appreciated!!

makermelissa, 2022-05-27:
It looks like it's failing `import gpiod` here: https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py#L6

However, you say you can run `import libgpiod` without issue. Sometimes if you run python with and without sudo it gets a different set of available libraries. Also, are you running inside of an env? This sounds like it may be something specific to your setup. Can you verify your `import gpiod` and `import board` tests are from the same python instance?

Andrew-xyz, 2022-05-27:
Tried a few more things with no luck. I'm getting this error with the gpiod.Chip object still:

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 83, in <module>
    from adafruit_blinka.board.beagleboard.beaglebone_ai import *
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/board/beagleboard/beaglebone_ai.py", line 5, in <module>
    from adafruit_blinka.microcontroller.dra74x import pin
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/dra74x/pin.py", line 10, in <module>
    P8_3 = Pin((0, 24))  # GPIO1_24 - GPIO_24
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 35, in __init__
    self._chip = gpiod.Chip(str(pin_id[0]), gpiod.Chip.OPEN_BY_NUMBER)
AttributeError: type object 'gpiod.Chip' has no attribute 'OPEN_BY_NUMBER'`

I found this very similar issue posted late last year:
https://webcache.googleusercontent.com/search?q=cache:53uIGYCWaIoJ:https://bytemeta.vip/repo/adafruit/Adafruit_Blinka/issues/497+&cd=2&hl=en&ct=clnk&gl=us&client=safari

Here, it was recommended to install libgpiod as a package as shown here:
https://learn.adafruit.com/adding-a-single-board-computer-to-blinka/software-setup#installing-the-libgpiod-as-a-package-3058813-3

I've followed the steps as shown here, as it had seemed to work for others, but no luck again unfortunately! I'm wondering if installing via the raspberry pi script may have caused issues, I'll try with a fresh Debian image and install as a package only to test.

Andrew-xyz, 2022-05-27:
That did the trick for me! So sorry to raise the issue. For anyone else looking to implement this specific solution:

Getting started:
https://beagleboard.org/getting-started

Started from bare bones Debian image available here:
https://debian.beagleboard.org/images/am57xx-debian-10.3-console.armhf-2020-04-06-1gb.img.xz
**NOTE: Be sure to edit /boot/uEnv.txt and un-comment the last line to enable the eMMC flasher**

Set up wifi via connmanctl utility:
https://www.fis.gatech.edu/how-to-configure-bbw-wifi/

`$ sudo apt update`
`$ sudo apt upgrade`
`$ sudo apt install -y python3 python3-pip python3-libgpiod`
`$ sudo pip3 install adafruit_blinka`

To test if you've done this properly:
`$ sudo python3`
`>>> import gpiod`
`>>> import board`

If these imports work, all set! Thanks to @makermelissa , all of this information is out there and has been answered by them already. I've just compiled it here for anyone else having the same issues.

closed 2022-05-27 by Andrew-xyz

makermelissa, 2022-05-27:
Awesome!