adafruit/Adafruit_Blinka #596: Pip install fails on Beaglebone Black

scirelli, 2022-07-14:
I found this [page](https://circuitpython.org/blinka/beaglebone_black) and it seemed like I could install Blinka on a BBB. When I tried I got the below errors.

Running:
- Python 3.10.4
- pip 22.1.2

It looks like it tries to install `RPi.GPIO` Raspberry Pi GPIO?

```
$ pip3 install Adafruit-Blinka
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting Adafruit-Blinka
  Downloading https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-8.0.2-py3-none-any.whl (231 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 231.2/231.2 kB 503.1 kB/s eta 0:00:00
Collecting pyftdi>=0.40.0
  Downloading https://www.piwheels.org/simple/pyftdi/pyftdi-0.54.0-py3-none-any.whl (144 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 144.8/144.8 kB 410.9 kB/s eta 0:00:00
Collecting rpi-ws281x>=4.0.0
  Downloading rpi_ws281x-4.3.4.tar.gz (64 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 64.2/64.2 kB 923.1 kB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Requirement already satisfied: Adafruit-PlatformDetect>=3.13.0 in /media/sdcard/home/cck/.pyenv/versions/3.10.4/lib/python3.10/site-packages (from Adafruit-Blinka) (3.24.1)
Collecting adafruit-circuitpython-typing
  Downloading https://www.piwheels.org/simple/adafruit-circuitpython-typing/adafruit_circuitpython_typing-1.7.1-py3-none-any.whl (9.2 kB)
Collecting Adafruit-PureIO>=1.1.7
  Downloading https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.9-py3-none-any.whl (12 kB)
Collecting sysv-ipc>=1.1.0
  Downloading sysv_ipc-1.1.0.tar.gz (99 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 99.4/99.4 kB 1.1 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting RPi.GPIO
  Downloading RPi.GPIO-0.7.1.tar.gz (29 kB)
  Preparing metadata (setup.py) ... done
Collecting pyusb!=1.2.0,>=1.0.0
  Downloading https://www.piwheels.org/simple/pyusb/pyusb-1.2.1-py3-none-any.whl (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.4/58.4 kB 356.4 kB/s eta 0:00:00
Collecting pyserial>=3.0
  Downloading https://www.piwheels.org/simple/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 156.9 kB/s eta 0:00:00
Collecting adafruit-circuitpython-busdevice
  Downloading https://www.piwheels.org/simple/adafruit-circuitpython-busdevice/adafruit_circuitpython_busdevice-5.1.10-py3-none-any.whl (7.3 kB)
Building wheels for collected packages: rpi-ws281x, sysv-ipc, RPi.GPIO
  Building wheel for rpi-ws281x (setup.py) ... done
  Created wheel for sysv-ipc: filename=sysv_ipc-1.1.0-cp310-cp310-linux_armv7l.whl size=55284 sha256=1fa4e788fe84346b127d3de42e213ba96bcedaf2ae167a44612bf8721a6a28c5
  Stored in directory: /media/sdcard/home/cck/.cache/pip/wheels/d0/44/ea/b9041c5c17d02075484f7745c4d2db8c9b79af9c2cf8f82a23
  Building wheel for RPi.GPIO (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [48 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-3.10
      creating build/lib.linux-armv7l-3.10/RPi
      copying RPi/__init__.py -> build/lib.linux-armv7l-3.10/RPi
      creating build/lib.linux-armv7l-3.10/RPi/GPIO
      copying RPi/GPIO/__init__.py -> build/lib.linux-armv7l-3.10/RPi/GPIO
      running build_ext
      building 'RPi._GPIO' extension
      creating build/temp.linux-armv7l-3.10
      creating build/temp.linux-armv7l-3.10/source
      clang -pthread -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/media/sdcard/home/cck/.pyenv/versions/3.10.4/include/python3.10 -c source/c_gpio.c -o build/temp.linux-armv7l-3.10/source/c_gp
io.o
      clang -pthread -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/media/sdcard/home/cck/.pyenv/versions/3.10.4/include/python3.10 -c source/common.c -o build/temp.linux-armv7l-3.10/source/comm
on.o
      clang -pthread -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/media/sdcard/home/cck/.pyenv/versions/3.10.4/include/python3.10 -c source/constants.c -o build/temp.linux-armv7l-3.10/source/c
onstants.o
[... 84 lines cut ...]
                  ^
      source/py_gpio.c:323:12: warning: implicit declaration of function 'setup_one' is invalid in C99 [-Wimplicit-function-declaration]
            if (!setup_one())
                 ^
      source/py_gpio.c:345:21: error: function definition is not allowed here
         int output(void) {
                          ^
      source/py_gpio.c:419:18: error: called object type 'PyObject *' (aka 'struct _object *') is not a function or function pointer
            if (!output())
                 ~~~~~~^
      source/py_gpio.c:475:18: error: called object type 'PyObject *' (aka 'struct _object *') is not a function or function pointer
            if (!output())
                 ~~~~~~^
      4 warnings and 5 errors generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> RPi.GPIO

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

scirelli, 2022-08-15:
@makermelissa could you point me in a direction to fix this? Maybe I can contribute a patch?

scirelli, 2022-08-15:
Is it as simple as being more specific about the architecture? https://github.com/adafruit/Adafruit_Blinka/blob/main/requirements.txt#L4

makermelissa, 2022-08-15:
Here's the setup.py where it determines if RPi.GPIO needs to be installed: https://github.com/adafruit/Adafruit_Blinka/blob/main/setup.py#L25-L37 Offhand, I'm really not sure why it would be acting this way. It's possible BBB added something to say it's compatible with Broadcom chips for some reason, but that seems unlikely. For other files specific to the BBB, the board file is at https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/board/beagleboard/beaglebone_black.py and the microcontroller files are here: https://github.com/adafruit/Adafruit_Blinka/tree/main/src/adafruit_blinka/microcontroller/am335x

scirelli, 2022-08-16:
I'm not sure what changed but this time the install worked. I guess 10th time's the charm?
```
pipenv install Adafruit-Blinka
Installing Adafruit-Blinka...
✔ Installation Succeeded
Pipfile.lock not found, creating...
Locking [dev-packages] dependencies...
Locking [packages] dependencies...
✔ Success!
Updated Pipfile.lock (98c947)!
Installing dependencies from Pipfile.lock (98c947)...
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 0/0 — 00:00:00
  ```
  
Testing fails though. I attempted to `import board`, it failed, #599. I'm going to attempt another install on a second BBB.

scirelli, 2022-08-16:
Install fails on the second BBB. I'm not sure what the difference is yet. They should be the exact same setup.

B1 (failed)
Python 3.10.4
pip 22.1.2 from /tmp/t/.venv/lib/python3.10/site-packages/pip (python 3.10)

B2 (succeeded)
Python 3.10.4
pip 22.1.2 from /tmp/t2/.venv/lib/python3.10/site-packages/pip (python 3.10)

```
pipenv install Adafruit-Blinka
Installing Adafruit-Blinka...
Error:  An error occurred while installing Adafruit-Blinka!
Error text: Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting adafruit-blinka 
  Downloading https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-8.2.0-py3-none-any.whl (234 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 234.3/234.3 kB 537.4 kB/s eta 0:00:00
Collecting adafruit-circuitpython-typing
  Downloading adafruit_circuitpython_typing-1.7.3-py3-none-any.whl (9.1 kB)
Collecting RPi.GPIO
  Using cached RPi.GPIO-0.7.1.tar.gz (29 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting Adafruit-PlatformDetect>=3.13.0
  Downloading https://www.piwheels.org/simple/adafruit-platformdetect/Adafruit_PlatformDetect-3.27.0-py3-none-any.whl (18 kB)
Collecting pyftdi>=0.40.0  
  Using cached https://www.piwheels.org/simple/pyftdi/pyftdi-0.54.0-py3-none-any.whl (144 kB)
Collecting sysv-ipc>=1.1.0 
  Using cached sysv_ipc-1.1.0-cp310-cp310-linux_armv7l.whl
Collecting Adafruit-PureIO>=1.1.7
  Using cached https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.9-py3-none-any.whl (12 kB)
Collecting rpi-ws281x>=4.0.0
  Using cached rpi_ws281x-4.3.4-cp310-cp310-linux_armv7l.whl
Collecting pyusb!=1.2.0,>=1.0.0
  Using cached https://www.piwheels.org/simple/pyusb/pyusb-1.2.1-py3-none-any.whl (58 kB)
Collecting pyserial>=3.0   
  Using cached https://www.piwheels.org/simple/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting adafruit-circuitpython-busdevice
  Downloading adafruit_circuitpython_busdevice-5.2.1-py3-none-any.whl (7.4 kB)
Collecting typing-extensions~=4.0
  Using cached https://www.piwheels.org/simple/typing-extensions/typing_extensions-4.3.0-py3-none-any.whl (25 kB)
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py): started
  Building wheel for RPi.GPIO (setup.py): finished with status 'error'
  Running setup.py clean for RPi.GPIO
Failed to build RPi.GPIO   
Installing collected packages: sysv-ipc, rpi-ws281x, RPi.GPIO, pyserial, typing-extensions, pyusb, Adafruit-PureIO, Adafruit-PlatformDetect, pyftdi, adafruit-circuitpython-busdevice, adafruit-circuitpython-typing, adafruit-blinka
  Running setup.py install for RPi.GPIO: started
  Running setup.py install for RPi.GPIO: finished with status 'error'
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [50 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-cpython-310
      creating build/lib.linux-armv7l-cpython-310/RPi
      copying RPi/__init__.py -> build/lib.linux-armv7l-cpython-310/RPi
      creating build/lib.linux-armv7l-cpython-310/RPi/GPIO
copying RPi/GPIO/__init__.py -> build/lib.linux-armv7l-cpython-310/RPi/GPIO
      warning: build_py: byte-compiling is disabled, skipping.
      
      running build_ext
      building 'RPi._GPIO' extension
      creating build/temp.linux-armv7l-cpython-310
      creating build/temp.linux-armv7l-cpython-310/source
[... 84 lines cut ...]
            if (!setup_one())
                 ^
      source/py_gpio.c:345:21: error: function definition is not allowed here
         int output(void) {
                          ^
      source/py_gpio.c:419:18: error: called object type 'PyObject *' (aka 'struct _object *') is not a function or function pointer
            if (!output())
                 ~~~~~~^
      source/py_gpio.c:475:18: error: called object type 'PyObject *' (aka 'struct _object *') is not a function or function pointer
            if (!output())
                 ~~~~~~^
      4 warnings and 5 errors generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> RPi.GPIO

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
✘ Installation Failed
```

makermelissa, 2022-08-16:
I'm glad you were able to get one working at least. My guess is you installed something it needed when playing around. Which Beaglebone images do you have installed on those?

scirelli, 2022-08-16:
Yeah, the board that does not work is the one I've been playing around with, and testing things. I must have installed something that is causing a problem.

But, even on the board where Blinka does install, attempting to use Blinka [fails](https://github.com/adafruit/Adafruit_Blinka/issues/599). 

Umm I can't remember now exactly which one it is. But I used one of the [eMMC flasher images](https://beagleboard.org/latest-images)
```
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

I guess a version of this
> Buster IoT (without graphical desktop) for [BeagleBone](https://beagleboard.org/bone) on-board eMMC flashing via microSD card
[AM3358 Debian 10.3 2020-04-06 4GB eMMC IoT Flasher ](https://debian.beagleboard.org/images/bone-eMMC-flasher-debian-10.3-iot-armhf-2020-04-06-4gb.img.xz)image for [BeagleBone](https://beagleboard.org/bone-original), [BeagleBone Black](https://beagleboard.org/black), [BeagleBone Black Wireless](https://beagleboard.org/black-wireless), [BeagleBone Blue](https://beagleboard.org/blue), [SeeedStudio BeagleBone Green](https://beagleboard.org/green), [SeeedStudio BeagleBone Green Wireless](https://beagleboard.org/green-wireless), [SanCloud BeagleBone Enhanced](https://beagleboard.org/enhanced), [element14 BeagleBone Black Industrial](https://beagleboard.org/e14-bbbi), [Arrow BeagleBone Black Industrial](https://beagleboard.org/arrowbbbi) and [Mentorel BeagleBone uSomIQ](https://beagleboard.org/usomiq) - [more info ](https://debian.beagleboard.org/)- sha256sum: e339459077b83f6458cb3432494954582aedad897b9f3b62fa390dfdb010a9df

scirelli, 2022-08-16:
A little progress. On the BBB where the installed failed I switched to using the system Python (3.7.3) and the install succeeded. I am going to uninstall 3.10.4 and reinstall it. See if that helps.

scirelli, 2022-08-16:
By manually installing `Adafruit_BBIO` I was able to run the previously failing tests from the [docs](https://learn.adafruit.com/circuitpython-essentials/circuitpython-pins-and-modules#import-board-3082637)

```
import board
dir(board)
```

scirelli, 2022-08-16:
I'm going to close this issue as it looks like it was just something messed up with my Python install.

closed 2022-08-16 by scirelli