adafruit/Adafruit_Blinka #599: Beaglebone Black: Import Board Fails with "No module named 'Adafruit_BBIO'"

scirelli, 2022-08-16:
While attempting to follow https://learn.adafruit.com/circuitpython-essentials/circuitpython-pins-and-modules#import-board-3082637

This error occurred: 
```
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/media/sdcard/home/cck/.local/share/virtualenvs/t-dpegpqp7/lib/python3.10/site-packages/board.py", line 62, in <module>
    from adafruit_blinka.board.beagleboard.beaglebone_black import *
  File "/media/sdcard/home/cck/.local/share/virtualenvs/t-dpegpqp7/lib/python3.10/site-packages/adafruit_blinka/board/beagleboard/beaglebone_black.py", line 5, in <module>
    from adafruit_blinka.microcontroller.am335x import pin
  File "/media/sdcard/home/cck/.local/share/virtualenvs/t-dpegpqp7/lib/python3.10/site-packages/adafruit_blinka/microcontroller/am335x/pin.py", line 5, in <module>
    from Adafruit_BBIO import GPIO
ModuleNotFoundError: No module named 'Adafruit_BBIO'
```

scirelli, 2022-08-16:
Manually installing Adafruit_BBIO
```
pip install Adafruit_BBIO
```
fixes the `ModuleNotFoundError`. I was able to run the test from the [docs](https://learn.adafruit.com/circuitpython-essentials/circuitpython-pins-and-modules#import-board-3082637)
```
import board
dir(board)
```

scirelli, 2022-08-16:
Running a [search](https://github.com/adafruit/Adafruit_Blinka/search?q=adafruit_bbio) for `adafruit_bbio`
There is one reference to it [pin.py line 5](https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/microcontroller/am335x/pin.py#L5).

Should it be added to this packages dependencies? Or something added to the README when installing to the beaglebone?

makermelissa, 2022-08-16:
> Manually installing Adafruit_BBIO
> 
> ```
> pip install Adafruit_BBIO
> ```
> 
> fixes the `ModuleNotFoundError`. I was able to run the test from the [docs](https://learn.adafruit.com/circuitpython-essentials/circuitpython-pins-and-modules#import-board-3082637)

Good find! Thank you. 

> Should it be added to this packages dependencies? Or something added to the README when installing to the beaglebone?

I think so, but only if we can do it for certain processors like we have now in setup.py for other boards and possibly requirements.txt. My concern is it isn't available for most boards, so I don't want to break it for those it's unavailable for.

aggieNick02, 2022-12-10:
I just tripped over this too. Unfortunately googling for the title of this bug doesn't even get one to this issue. It was only upon using github search on this repo for "Adafruit_BBIO" that I landed here.

Thanks for filing the issue and explaining. Without the issue I would have been confused for a bit longer. :-)

makermelissa, 2023-09-15:
I just set up a fresh BBB today and didn't run into this issue, but in order to help prevent it from occurring, in #714 I added it to `setup.py` and improved the error message if it isn't found.

PR #714 (2023-09-15): Add better error message and explicity add to setup
Fixes #599.
closed 2023-09-15 by makermelissa