adafruit/Adafruit_Blinka #600: Tests from Docs Fail

scirelli, 2022-08-16:
Maybe this is just a documentation update? I'm not seeing any testing module installed.

Example from [docs](https://docs.circuitpython.org/projects/blinka/en/latest/index.html#usage-example)
```
from testing import test_module_name
test_module_name("testing.universal.digitalio")
```
 Error
```
import testing
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'testing'
```

makermelissa, 2022-08-17:
Yeah, perhaps some more detailed or specific instructions are in order.

I think I tested this successfully on a Pi or something and I think maybe you had to be in a specific directory.

scirelli, 2022-08-17:
My thought was these tests were for the devs, since the docs test folder structure matches the source code. So I looked into the install `.venv/lib/python3.10/site-packages/adafruit_*` and didn't see the tests.

makermelissa, 2022-08-17:
Yeah, they basically are, but there's been a lot added to Blinka without any of the tests being updated (or even used AFAIK) so they aren't very useful at the moment.

PR #710 (2023-09-13): Update Usage Example in Readme
Fixes #600. I replaced the usage example with something more practical.
closed 2023-09-13 by makermelissa