adafruit/Adafruit_Blinka #602: Adafruit_BBIO cleanup Function not Called

scirelli-c1, 2022-08-17:
I noticed the Pin class uses Adafruit_BBIO.GPIO but doesn't call the `cleanup` function, or surface the clean up function in the [pin.py Pin class](https://github.com/adafruit/Adafruit_Blinka/blob/ef2657eaa1caf3262c6bbe1f2df5b6ed08f72162/src/adafruit_blinka/microcontroller/am335x/pin.py#L8-L65) for the user to call. It's recommend to be called in the [GPIO docs](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/docs/GPIO.rst).

I guess this issue is also a question; does it need to be called since BBIO.GPIO only recommends it?

```
.. function:: cleanup()

   Clean up by resetting all GPIO channels that have been used by
   the application to :data:`IN` with no pullup/pulldown and no event
   detection.

   :note: It's recommended that you call this function upon exiting your
       application.
```

makermelissa, 2022-08-17:
Good question. The beagleboard code is old and has been around long before I touched Blinka. It's possible either it was causing an issue and not added. We can always try adding it and remove it if there are issues that crop up, but it's one of those things that nobody has said they had any problems from.

makermelissa, 2023-09-15:
Since it is only recommended, but not causing any issues, I'm going to close this.

closed 2023-09-15 by makermelissa