adafruit/Adafruit_Blinka #609: Possibly split up MicroPython and Linux stuff a bit more

makermelissa, 2022-09-06:
In the future, I think splitting Blinka up a bit more internally between the Linux and MicroPython stuff would be helpful with fixing a few issues:

1. With adding typing info, Blinka has been super touchy because it's supposed to work in such a wide variety of environments.
2. MicroPython size issues. Splitting it up could make it easier to have something more compact that could be extracted to work on MicroPython. For instance, `busio.py` has grown quite a bit and I've heard complaints about people running out of space when they have more than a couple CircuitPython libraries in use. I'm sure there are quite a few other modules like `microcontroller/pin.py` that are used on MP and are only large due to the huge if/elif statements.
3. This could be a better alternative to the related issue #447 or it could be part of it.

makermelissa, 2023-03-24:
As the number of supported boards have grown since I opened this, the MicroPython size issues have become more of an issue for some people. It may be worth moving the huge board if/else statement into another file and if the board is determined not to be MP, then import that and go through the boards. Perhaps something similar could be done with busio, except in that case the MP boards are currently checked last.

makermelissa, 2023-03-28:
It's still pretty popular with users that want to run MP on the Pico and use the CP libraries. This was more about making it easier for these users as opposed to taking away that functionality.

anguselheimStudios, 2023-05-05:
Maybe split it entirely and make them each they're own thing? That way the MP people don't have to bothered with the unwieldy effects of  SBC/SOC maker's lack of standardization at all. I was also thinking you could do something with fstrings and the board's compatible and the file names for the board definitions to clear out those if/then's. 

I'm willing to try the latter, but Disclamner: 

I only have 2 less-than-popular boards to test with, and blinka only works on 1 of them right now (as per aforementioned SBC-maker shenanigans).

I'm not quite a n00b, but by no means a pro.

I have a more-than-full-time job, so I don't work fast.

Would you like me to take a crack at it?

makermelissa, 2026-03-12:
This is improved by https://github.com/adafruit/Adafruit_Blinka/pull/1040, which reduces the file size considerably.

makermelissa, 2026-04-22:
I think I'm going to close because it solves many of the issues with #1040.