adafruit/Adafruit_Blinka #611: usb_hid mouse  Raspberry Pi Zero 2 W

Med-JH, 2022-09-07:
I cant get mouse emulation working, 

It is either becoming stuck or gives an error.
If I follow [this example](https://stackoverflow.com/questions/66393757/using-pi-zero-w-to-simulate-hid-keyboard-and-mouse) I can send keyboard and mouse input. So the hardware and OS should be ok.

Thank you

Error variant:
```
import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.mouse import Mouse
from adafruit_hid.keycode import Keycode
usb_hid.enable((usb_hid.Device.MOUSE,))
usb_hid.enable((usb_hid.Device.MOUSE,))
```
Error:
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/dist-packages/adafruit_hid/mouse.py", line 51, in __init__
>     self._send_no_move()
>   File "/usr/local/lib/python3.9/dist-packages/adafruit_hid/mouse.py", line 148, in _send_no_move
>     self._mouse_device.send_report(self.report)
>   File "/usr/local/lib/python3.9/dist-packages/usb_hid.py", line 66, in send_report
>     fd.write(report)
> BlockingIOError: [Errno 11] write could not complete without blocking
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/usr/local/lib/python3.9/dist-packages/adafruit_hid/mouse.py", line 54, in __init__
>     self._send_no_move()
>   File "/usr/local/lib/python3.9/dist-packages/adafruit_hid/mouse.py", line 148, in _send_no_move
>     self._mouse_device.send_report(self.report)
>   File "/usr/local/lib/python3.9/dist-packages/usb_hid.py", line 66, in send_report
>     fd.write(report)
> BlockingIOError: [Errno 11] write could not complete without blocking
> 

stuck variant:
```
import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.mouse import Mouse
from adafruit_hid.keycode import Keycode
usb_hid.enable((usb_hid.Device.KEYBOARD, usb_hid.Device.MOUSE),)
mouse = Mouse(usb_hid.devices)
```
Spawns a keyboard on a connected Windows PC but no mouse. Instead of the mouse I get an failed HID device with Error:
> This device cannot start. (Code 10)
> The I/O request was canceled.

mouse = Mouse(usb_hid.devices) never finishes

if I use

`usb_hid.enable((usb_hid.Device.KEYBOARD, usb_hid.Device.MOUSE),boot_device=2)`
I get a mouse but no keyboard

my preparation 

lsb_release -a
```
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
```

boot/config.txt
```
...
dtoverlay=dwc2
enable_uart=1
dtoverlay=tc358743
...
```
pip list
```
Adafruit-Blinka                  8.2.1
adafruit-circuitpython-busdevice 5.2.3
adafruit-circuitpython-hid       5.3.2
adafruit-circuitpython-requests  1.12.8
adafruit-circuitpython-typing    1.8.1
Adafruit-PlatformDetect          3.27.2
Adafruit-PureIO                  1.1.9
adafruit-python-shell            1.3.3
```

makermelissa, 2022-09-07:
I noticed the example you linked to was for the original Raspberry Pi Zero W and you have the Pi Zero 2 W. Is it possible some functionality may have been removed?

Med-JH, 2022-09-08:
I dont think that there is anything missing on the Pi Zero 2 becausse the example works, a "boot" mouse works and I also tested pikvm which has the functionality. 
My guess is, that it is something with the reports, but I cant figure it out.

makermelissa, 2022-09-15:
Good find. Like 99.9% of this was written by a community contributor, so I'm not really familiar with this module at all without really diving in. If you'd like to try and get it working, I'd be happy to review a PR.

makermelissa, 2026-10-29:
Fixed by #1154. Confirmed testing on both Mac and Windows.

closed 2026-10-29 by makermelissa