adafruit/Adafruit_Blinka #617: Libre AML-S905X-CC (LePotato) New Board Request

Snakn1, 2022-09-20:
Can you please add the  Libre AML-S905X-CC (LePotato) the pin info can be found at [https://github.com/libre-computer-project/libretech-wiring-tool/blob/master/libre-computer/aml-s905x-cc/gpio.map](https://github.com/libre-computer-project/libretech-wiring-tool/blob/master/libre-computer/aml-s905x-cc/gpio.map)

or 
[https://docs.google.com/spreadsheets/d/1U3z0Gb8HUEfCIMkvqzmhMpJfzRqjPXq7mFLC-hvbKlE/edit?usp=sharing](https://docs.google.com/spreadsheets/d/1U3z0Gb8HUEfCIMkvqzmhMpJfzRqjPXq7mFLC-hvbKlE/edit?usp=sharing)

makermelissa, 2022-09-20:
If you'd like to try, here are the guides that go over how to add a board to Blinka:
https://learn.adafruit.com/adding-a-single-board-computer-to-platformdetect-for-blinka
https://learn.adafruit.com/adding-a-single-board-computer-to-blinka

Snakn1, 2022-09-20:
I am sorry I am not sure how to reply so if I need to reply on GitHub then
let me know.

I can not detect the cpu or the board type per your directions
using detect.py.

I am running "ubuntu-22.04.1-preinstalled-desktop-arm64+aml-s905x-cc"

Given the information I have, what is the next step?

sudo python3 detect.py

Chip id:  None
Board id:  None

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Board? False
Is this a Giant Board? False
Is this a Coral Dev Board? False
Is this a Coral Dev Board Mini? False
Is this a MaaXBoard? False
Is this a MaaXBoard Mini? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a NanoPi board? False
Is this a Khadas VIM3 board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? False
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an ASUS Tinker Board? False
Is this an STM32MP1 Board? False
Is this an OS environment variable special case? False

more /proc/cpuinfo

processor       : 0
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

/proc$ cat version
Linux version 5.19.9-03540-gea78233c0133 ***@***.***)
(gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0, GNU ld (GNU Binutils for Ubuntu)
2.38) #1 SMP PREEMPT Sat Sep 17 13:30:09 UTC 2022

 /proc$ cat devices

I: Bus=001e Vendor=0000 Product=0000 Version=0001
N: Name="meson_ao_cec"
P: Phys=meson_ao_cec/input0
S: Sysfs=/devices/platform/soc/c8100000.bus/c8100100.cec/rc/rc0/input0
U: Uniq=
H: Handlers=kbd event0
B: PROP=20
B: EV=100017
B: KEY=ffffc000000000 3ff 0 400000320fc200 40830c900000000 0 210300
49d2c040ec00 1e378000000000 8010000010000000
B: REL=3
B: MSC=10

I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="meson-ir"
P: Phys=meson-ir/input0
S: Sysfs=/devices/platform/soc/c8100000.bus/c8100580.ir/rc/rc1/input1
U: Uniq=
H: Handlers=kbd event1
B: PROP=20
B: EV=100017
B: KEY=1000000 0 0
B: REL=3
B: MSC=10

from the /proc/device-tree/
cat model
ibre Computer AML-S905X-CC^@

***@***.***:/proc/device-tree$ ls

'#address-cells'    emmc-pwrseq        psci                   smbios
 aliases            firmware           regulator-hdmi-5v      soc
 analog-amplifier   hdmi-connector     regulator-vcc_3v3      sound
 arm-pmu            interrupt-parent   regulator-vcc-card     __symbols__
 chosen             leds               regulator-vddio_ao18   thermal-zones
 compatible         ***@***.***           regulator-vddio_boot   timer
 cpus               model              reserved-memory        xtal-clk
 cvbs-connector     name               scpi
 efuse              opp-table         '#size-cells'

proc$ ls

1     1303  1505  230   390   613   7348  8183    buddyinfo    meminfo
10    1319  1539  231   394   614   7362  8190    bus          misc
1030  1330  1551  232   395   615   74    8194    cgroups      modules
1034  1332  1570  2321  397   616   740   8203    cmdline      mounts
1042  1348  16    233   4     618   749   8211    config.gz    mtd
11    1349  1638  234   40    624   7490  8212    consoles     net
1142  1351  17    235   400   625   76    8213    cpuinfo      pagetypeinfo
12    1352  171   236   41    626   760   8214    crypto       partitions
1209  1353  1714  26    42    627   7608  8215    devices      self
1213  1356  172   268   43    628   7620  822     device-tree  slabinfo
1221  1357  173   269   44    629   7661  836     diskstats    softirqs
1227  1363  1753  27    45    63    77    837     driver       stat
1232  1365  1754  28    46    632   78    840     execdomains  swaps
1236  1369  18    3     47    668   796   856     fb           sys
1242  1372  184   303   48    68    797   858     filesystems  sysrq-trigger
1246  1373  2     31    49    692   8041  861     fs           sysvipc
1249  1376  21    32    5     697   8045  864     interrupts   thread-self
1259  1379  219   3260  50    6979  8062  869     iomem        timer_list
1263  1383  22    329   53    701   8071  870     ioports      tty
1267  1384  221   353   54    7026  8105  881     irq          uptime
127   14    222   37    55    711   813   885     kallsyms     version
1275  1402  223   38    56    7170  8133  891     keys         vmallocinfo
1278  1429  224   381   5942  7171  8139  9       key-users    vmstat
128   1433  225   382   597   7172  8140  915     kmsg         zoneinfo
1292  1438  226   383   598   72    8141  937     kpagecgroup
1295  144   227   384   6040  7250  8143  947     kpagecount
13    1444  228   386   605   7268  8144  976     kpageflags
130   1462  229   387   606   73    818   990     loadavg
1301  15    23    39    607   7319  8182  asound  locks

Snakn1, 2022-09-22:
Please help. Per the board support team" how can I modify your code to work to detect this bd?

Raspberry Pi does some non-standard manipulation of /proc/cpuinfo in their Linux. This method of detection is not supported by anything outside of the Raspberry Pi and Raspbian combination. We recommend using industry standard DMI interface sysfs entries instead of the non-standard Pi method.

/sys/class/dmi/id/board_name
/sys/class/dmi/id/board_vendor

makermelissa, 2022-09-22:
It looks like it might have an amlogic S905X chip judging by the board name. There are a number of other boards that are detecting similar amlogic chips (S905, S905X3, S905Y2, etc.) I would try and do something similar to those boards. It looks like their checking for values in /proc/device-tree/compatible. See https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/main/adafruit_platformdetect/chip.py#L291-L303 for where similar boards are being detected.

makermelissa, 2022-10-17:
Fixed via #620.

closed 2022-10-17 by makermelissa