adafruit/Adafruit_Blinka #622: I was using rev 1.4 and had no problems but recently bought some new 4b raspberry pi and I guess its 1.5 and now my clone os is throwing this error.

nolansingroy, 2022-10-17:
I was using rev 1.4 and had no problems but recently bought some new 4b raspberry pi and I guess its 1.5 and now my clone os is throwing this error. 
[image]

cat /proc/cpuinfo
[image]

  
 My cloned os using the adafruit plugin for neopixel was working on this version of raspberry pi 
Hardware	: BCM2711
Revision	: c03114
Serial		: 100000005a9870db
Model		: Raspberry Pi 4 Model B Rev 1.4
 

 **Please any advice or help would be greatly appreciated!**

__Originally posted by @nolansingroy in https://github.com/adafruit/Adafruit_Blinka/issues/66#issuecomment-1280200150__

closed 2022-10-17 by makermelissa

makermelissa, 2022-10-17:
Feel free to re-open this if that doesn't solve the issue.

nolansingroy, 2022-10-27:
@makermelissa the above was installed and it stopped saying board not support however when I run the adafruit blinka neopixel code I get 
[image]