adafruit/Adafruit_Blinka #625: RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)

nolansingroy, 2022-10-27:
@makermelissa the above was installed and it stopped saying board not support however when I run the adafruit blinka neopixel code I get 
[image]

_Originally posted by @nolansingroy in https://github.com/adafruit/Adafruit_Blinka/issues/622#issuecomment-1293991294_

makermelissa, 2022-10-27:
Ok, it sounds like an issue with the `rpi_ws281x` dependency. Make sure you're running the latest version with 
`pip3 install --upgrade rpi_ws281x`. If that doesn't work, I'd recommend filing an issue with https://github.com/rpi-ws281x/rpi-ws281x-python/

nolansingroy, 2022-10-27:
I originally `sudo pip3 install rpi_ws281x adafruit-circuitpython-neopixel` but it did not work until I tried the command @makermelissa mentioned above -- for others reading this when they get their rev 1.5 pi 4 board -- this was the fix.

closed 2022-10-27 by makermelissa