adafruit/Adafruit_Blinka #639: pulseio library is broken on Apple Silicon hosts

mobigroup, 2022-12-10:
I've checked two boards on Apple Silicon platform:
- [Adafruit MCP2221A Breakout](https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221) - pulseio is not supported. Hmm, why not?
- [u2if firmware on RPI Pico](https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-raspberry-pi-pico?view=all) - pulseio doesn't work while rip pico obviously supports it! Below I'm trying to fix the issue but the library is broken because it includes invalid binaries.

1. Fix /opt/homebrew/lib/python3.10/site-packages/pulseio.py
```
...
if detector.board.any_raspberry_pi or detector.board.pico_u2if:
    from adafruit_blinka.microcontroller.bcm283x.pulseio.PulseIn import PulseIn
...
```
2. Load the library and try to use it
```
import pulseio
pulses = pulseio.PulseIn(board.GP0, maxlen=200, idle_state=True)
OSError: [Errno 8] Exec format error: '/opt/homebrew/lib/python3.10/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein64'
```
Opa - the library is broken completely. So it doesn't work for any board.

Neradoc, 2022-12-10:
> * Hmm, why not?

Because it's not supported. I was not able to find documentation showing that the MCP2221A can count pulses and how to do it, but maybe I'm missing something. At best you can count using digitalio, but that's only able to count a pretty slow signal comparatively.

> * pulseio doesn't work while rip pico obviously supports it

I don't see support for reading pulses in the u2if firmware:
https://github.com/execuc/u2if#implemented-interfaces

What blinka is doing with U2IF is communicating via USB with the pico, using the available features of the U2IF firmware. Some might be in U2IF but not implemented in blinka, and it's worth looking into it then, but I don't think it's the case here. If I'm wrong, we can look into adding it.

You could also ask about adding those features to U2IF, but I don't know if the project is active.

> adafruit_blinka.microcontroller.bcm283x

`bcm283x` is the Raspberry Pi Micro Processor. That's for when Blinka is installed on a Raspberry Pi Single Board Computer and uses its features to communicate with the outside using its own pins. It doesn't and cannot exist on Mac (which don't have a pin header and are not SBCs). The names can be confusing, but I'm talking about the Raspberry Pis that run linux: Pi Zero, 1, 2, 3, 4...

You might be better off running dedicated software on the board (using Circuitpython for example) and communicate with it using pyserial if you need something like pulseio.

makermelissa, 2022-12-14:
> * pulseio is not supported. Hmm, why not?

See https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-mcp2221/faq-troubleshooting#faq-3030918 for more info.

> * pulseio doesn't work while rip pico obviously supports it! Below I'm trying to fix the issue but the library is broken because it includes invalid binaries.

@caternuson wrote this. He'll have to speak more about why it isn't supported in u2if. 

Alternatively a possible option (though probably not ideal) is maybe you could install CircuitPython to use pwmio for now. If you need the data on your computer, you could try using remote procedure calls to shuttle information back and forth. I wrote a guide on this using the MacroPad and python code to send messages via MQTT in https://learn.adafruit.com/macropad-remote-procedure-calls-over-usb-to-control-home-assistant/host-computer-code, but it could be adapted for more general use.

caternuson, 2022-12-14:
> @caternuson wrote this. He'll have to speak more about why it isn't supported in u2if.

See @Neradoc 's comment above. u2if does not have support for pulseio.

The u2if firmware is 3rd party.

makermelissa, 2022-12-14:
> > @caternuson wrote this. He'll have to speak more about why it isn't supported in u2if.
> 
> See @Neradoc 's comment above. u2if does not have support for pulseio.
> 
> The u2if firmware is 3rd party.

Ah, I didn't realize. Closing here for now. Perhaps somebody could submit a PR to https://github.com/adafruit/u2if to add pwmio support.

closed 2022-12-14 by makermelissa

caternuson, 2022-12-14:
Ideally the PR would be submitted to:
https://github.com/execuc/u2if

The Adafruit repo is a fork. It was forked to allowing PRing changes back upstream to execuc. I don't think we'd want to maintain a separate fork.