adafruit/Adafruit_Blinka #643: Does not work with MicroPython and Raspberry Pi Pico

cbmeeks, 2023-01-03:
I cannot get even the simplest program to build using the `usb_hid` module and a Raspberry Pi Pico.

I have uploaded the Adafruit-Blinka and other dependencies to the Pico.  But even this simple program fails:

    import time

    import board
    import digitalio
    import usb_hid

Removing 'usb_hid' builds fine (granted, it doesn't actually DO anything but no errors).

The error reported is:

    Traceback (most recent call last):
      File "<stdin>", line 5, in <module>
      File "/lib/usb_hid.py", line 14, in <module>
    ImportError: no module named 'typing'

When I try to install the typing module, Thonny complains it isn't a MicroPython module.  Installing it anyway fails when I try and run because 'typing' is simply too large to fit on the Pico from what I understand.

Any ideas on how to fix this?

Neradoc, 2023-01-03:
USB HID in Blinka is not implemented for Micropython, it's only implemented for Raspberry Pi OS, using gadget mode (available on the Pi Zero and Pi 4), which is why it doesn't bother to try/except the typing import. (Micropython ignores type hints).

For it to be supported on MP it would require HID support at the Micropython level, and implementing a version of the module that uses it. I believe it is an option in Micropython to compile for the pico with complete USB support, but maybe HID is only supported on the pyboard.
https://docs.micropython.org/en/latest/library/pyb.USB_HID.html

cbmeeks, 2023-01-03:
Oh, I was going by this tutorial and just noticed he is using Circuit Python and not MicroPython.

https://tutorial.cytron.io/2021/02/16/keyboard-emulator-using-raspberry-pi-pico-maker-pi-pico-and-circuitpython/

So I guess that's my mistake the whole time.  I'm new to Python on these devices.  :-)

cbmeeks, 2023-01-04:
Example works with CircuitPython.  So I guess this can close.
Thanks.

closed 2023-01-04 by cbmeeks