adafruit/Adafruit_Blinka #645: No such file or directory: adafruit-blinka/functions/hid.usb1/dev

aavendano, 2023-01-16:
Hi, 
I am using an RPI B 3 and Raspberry Pi OS. I am trying to build a remote keyboard using usb_hid. But I keep getting the error below.

`FileNotFoundError: [Errno 2] No such file or directory: '/sys/kernel/config/usb_gadget/adafruit-blinka/functions/hid.usb1/dev'`

It seems that I am missing some controllers. But I can not figure it out what or from where I should get it. Please, I would appreciate your help

Thanks!

Neradoc, 2023-01-16:
What is your code ? I believe that you need to call something like:
```py
import usb_hid
usb_hid.enable((usb_hid.Device.KEYBOARD,))
```
However, note that the only Raspberry Pis that support gadget mode are the Pi Zero (on the data USB port) and the Pi 4 (on the USB-C port), all the other Pis (and ports) go through an on-board hub, and cannot act as a keyboard.

aavendano, 2023-01-16:
I added a line to enable the device, as suggested. So my code now is as below:

`

import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS

usb_hid.enable((usb_hid.Device.KEYBOARD, ))

keyboard = Keyboard(usb_hid.Device.KEYBOARD)
keyboard_layout = KeyboardLayoutUS(keyboard)
key = "test keyboard"
keyboard_layout.write(key)
`
However, the error now is:

File "/home/alejandro/virtualkb/testkb.py", line 8, in <module>
    usb_hid.enable((device, ))
  File "/usr/local/lib/python3.9/dist-packages/usb_hid.py", line 725, in enable
    Path("%s/UDC" % this.gadget_root).write_text("%s" % udc.name, encoding="utf-8")
  File "/usr/lib/python3.9/pathlib.py", line 1275, in write_text
    return f.write(data)
OSError: [Errno 16] Device or resource busy

I was under the impression that the Rpi 3 allows keyboard functionalities by changing the boot config. But I may be wrong. Please, clarify

closed 2023-01-20 by aavendano