adafruit/Adafruit_Blinka #648: Blinka is not installed in active directory

RnoldR, 2023-01-19:
I installed blinka according to the instructions fount at: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi. Next I tried to run the `blinkatest.py` script that can be found in the URL above. This crashes at the first line (`import board`) because board cannot be found. 

When I run the test as `sudo python blinkatest.py` it works correctly. The same when I disable anaconda initialization in my .bashrc script and run as `python blinkatest.py`. So I figure the library scripts are not installed in the environment libraries but in some global library. Is this the expected behavior? If so, is there a workaround? Or should I use adafruit products without blinka?

I have a raspberry 4 with the latest 64-bit version of raspberry os and the latest version of anaconda (22.9.0) for arm. I use environments because projects may have conflicting demands.

tekktrik, 2023-01-20:
I think conda may have it's own package location.  Are you activating the Anaconda virtual environment before pip installing>?

RnoldR, 2023-01-20:
I activated the anaconda environment before pip by installing. The way Anaconda uses its virtual environments is the same like venv and is recognised by pip. What is atypical for the Blinka installing is the use of sudo. As far as I know that implies root takes over the install process and that probably the libraries land on a non-standard spot for the user (but probably standard for user root).

tekktrik, 2023-01-20:
You could try `which pip` and `sudo which pip` to see if it's using a different `pip`.

makermelissa, 2023-01-23:
What happens when you try installing Blinka without using sudo by typing `pip3 install Adafruit-Blinka`?

anguselheimStudios, 2023-05-05:
Using sudo with pip installs a package globally when not being used with a virtual environment. Not sure how it would behave with conda.

makermelissa, 2023-09-08:
Feel free to reopen this if needed. I know that Blinka can be installed without using sudo by running `pip3 install --upgrade adafruit-blinka`. The install script on the page you linked to has you run needs to be run as sudo because it sets things like enabling I2C and SPI and running apt-get, which need sudo. However, when it runs the pip command, sudo is not used (see https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/main/raspi-blinka.py#L83). To install in a virtual environment you should install with pip only. I have added an alert to the learn guide to indicate this.

closed 2023-09-08 by makermelissa