adafruit/Adafruit_Blinka #652: Neopixel Support for AMLogic Boards specifically S905X

Iainrach, 2023-02-08:
Trying to use the great Neopixel library on a Libre board but it seems like it is not supported. Any thoughts on providing this ?

It appears for the bcm283x boards you utilize the _rpi_ws281x library.

Does this mean it would be pretty difficult to get support for the Libre boards ?

makermelissa, 2023-03-15:
Do you know of a ws2812 library that runs on the libre?

Iainrach, 2023-03-16:
Not so far but I can drive a Max 7219 display over the SPI bus via Blinka
so perhaps this could work ??

Overview | CircuitPython NeoPixel Library Using SPI | Adafruit Learning
System
<https://learn.adafruit.com/circuitpython-neopixels-using-spi/overview>

makermelissa, 2023-03-16:
Yeah, it's a good alternative. Yeah, if there's no existing library, it can get pretty difficult to add.

makermelissa, 2023-03-16:
Also, I think PWM support is a requirement to add to other GPIO pins and most boards don't have it implemented and if they do, it often hasn't been added to Blinka.