adafruit/Adafruit_Blinka #656: Pin numbers for Rock 4C Plus not available

MathijsNL, 2023-02-26:
I am using the library on a Rock 4C Plus. I am trying to use GPIO 73 and 74 (pin 31 and 29) 

```
>>> board.board_id
'ROCK_PI_4C+'
```

Pin assignment:
```
outPin = board.B1
AttributeError: module 'board' has no attribute 'B1'
```

As I understand the mapping is available here:
https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/microcontroller/rockchip/rk3399/pin.py#L24-L25

My assumption is that I need to use pin board.B1 and board.B2, but they don't show up. All available pins are shown below:

```
board.ADC_IN0   board.D27       board.D8        board.SDA7
board.CS        board.D28       board.I2C(      board.SPI(
board.D10       board.D29       board.MISO      board.UART2_RX
board.D11       board.D3        board.MOSI      board.UART2_TX
board.D12       board.D31       board.PWM0      board.UART4_RX
board.D13       board.D32       board.SCK       board.UART4_TX
board.D15       board.D33       board.SCL       board.UART_RX
board.D16       board.D35       board.SCL2      board.UART_TX
board.D18       board.D36       board.SCL6      board.ap_board
board.D19       board.D37       board.SCL7      board.board_id
board.D21       board.D38       board.SCLK      board.detector
board.D22       board.D40       board.SDA       board.pin
board.D23       board.D5        board.SDA2      board.sys
board.D24       board.D7        board.SDA6

board.pin
<module 'adafruit_blinka.microcontroller.rockchip.rk3399_t.pin' from '/home/rock/.local/lib/python3.8/site-packages/adafruit_blinka/microcontroller/rockchip/rk3399_t/pin.py'
```

@wgroeneveld Tagging you if you don't mind. Do you happen to know what the problem is?

makermelissa, 2023-02-27:
It's likely related to this: https://github.com/adafruit/Adafruit_Blinka/pull/651#issuecomment-1439630083, meaning the pinout is was added incorrectly for your board. I'm guessing your board is a production board (the one that was used to figure out the pinout wasn't), so maybe with your help, we could get it working (hopefully with both versions). If we can't get it working with both versions, I'd prefer the production board be the one that works.

MathijsNL, 2023-02-27:
I can confirm my 4C+ is version 1.2.

Will see what I can do about the mapping and board version detection.

makermelissa, 2023-02-27:
My best tips are in this guide: https://learn.adafruit.com/adding-a-single-board-computer-to-blinka

MathijsNL, 2023-02-27:
Just documenting things in here for reference.

> sudo gpiodetect
```
gpiochip0 [gpio0] (32 lines)
gpiochip1 [gpio1] (32 lines)
gpiochip2 [gpio2] (32 lines)
gpiochip3 [gpio3] (32 lines)
gpiochip4 [gpio4] (32 lines)
gpiochip5 [rk817-gpio] (1 lines)

```

> sudo gpioinfo
```

gpiochip0 - 32 lines:
        line   0:      unnamed       unused   input  active-high
        line   1:      unnamed       unused   input  active-high
        line   2:      unnamed       unused   input  active-high
        line   3:      unnamed       unused   input  active-high
        line   4:      unnamed "host-wakeup" input active-high [used]
        line   5:      unnamed       unused   input  active-high
        line   6:      unnamed       unused   input  active-high
        line   7:      unnamed       unused   input  active-high
        line   8:      unnamed       unused   input  active-high
        line   9:      unnamed   "shutdown"  output  active-high [used]
        line  10:      unnamed      "reset"  output   active-low [used]
        line  11:      unnamed       unused   input  active-high
        line  12:      unnamed       unused   input  active-high
        line  13:      unnamed       unused   input  active-high
        line  14:      unnamed       unused   input  active-high
        line  15:      unnamed       unused   input  active-high
        line  16:      unnamed       unused   input  active-high
        line  17:      unnamed       unused   input  active-high
        line  18:      unnamed       unused   input  active-high
        line  19:      unnamed       unused   input  active-high
        line  20:      unnamed       unused   input  active-high
        line  21:      unnamed       unused   input  active-high
        line  22:      unnamed       unused   input  active-high
        line  23:      unnamed       unused   input  active-high
        line  24:      unnamed       unused   input  active-high
        line  25:      unnamed       unused   input  active-high
        line  26:      unnamed       unused   input  active-high
        line  27:      unnamed       unused   input  active-high
        line  28:      unnamed       unused   input  active-high
        line  29:      unnamed       unused   input  active-high
        line  30:      unnamed       unused   input  active-high
        line  31:      unnamed       unused   input  active-high
gpiochip1 - 32 lines:
        line   0:      unnamed       unused   input  active-high
        line   1:      unnamed       unused  output  active-high
        line   2:      unnamed       unused   input  active-high
        line   3:      unnamed "vbus-typec"  output  active-high [used]
        line   4:      unnamed       unused   input  active-high
        line   5:      unnamed       unused   input  active-high
        line   6:      unnamed       unused   input  active-high
        line   7:      unnamed       unused   input  active-high
        line   8:      unnamed       unused   input  active-high
        line   9:      unnamed       unused   input  active-high
        line  10:      unnamed       unused   input  active-high
        line  11:      unnamed       unused   input  active-high
        line  12:      unnamed       unused   input  active-high
        line  13:      unnamed       unused   input  active-high
        line  14:      unnamed       "vsel"  output  active-high [used]
        line  15:      unnamed       unused   input  active-high
        line  16:      unnamed       unused   input  active-high
        line  17:      unnamed       "vsel"  output  active-high [used]
        line  18:      unnamed       unused   input  active-high
        line  19:      unnamed       unused   input  active-high
[... 90 lines cut ...]
        line  11:      unnamed       unused   input  active-high
        line  12:      unnamed       unused   input  active-high
        line  13:      unnamed       unused   input  active-high
        line  14:      unnamed       unused   input  active-high
        line  15:      unnamed       unused   input  active-high
        line  16:      unnamed       unused   input  active-high
        line  17:      unnamed       unused   input  active-high
        line  18:      unnamed       unused   input  active-high
        line  19:      unnamed       unused   input  active-high
        line  20:      unnamed       unused   input  active-high
        line  21:      unnamed       unused   input  active-high
        line  22:      unnamed       unused   input  active-high
        line  23:      unnamed       unused   input  active-high
        line  24:      unnamed       unused   input  active-high
        line  25:      unnamed        "hpd"   input   active-low [used]
        line  26:      unnamed       unused   input  active-high
        line  27:      unnamed       unused  output  active-high
        line  28:      unnamed       unused   input  active-high
        line  29:      unnamed       unused   input  active-high
        line  30:      unnamed       unused   input  active-high
        line  31:      unnamed       unused   input  active-high
gpiochip5 - 1 lines:
        line   0:      unnamed       unused  output  active-high

```

MathijsNL, 2023-02-28:
Almost there I guess, just a few questions remaining:

- How to do a proper board detection? Is there a way to get that somewhere from the system?
- Is it correct that SPI Mosi should be mapped to the TX?

If there is a way to distinguish between ROCK_PI_4_C_PLUS < 1.2 and ROCK_PI_4_C_PLUS > 1.2 I think this is probably the place to add it?

```
elif board_id == ap_board.ROCK_PI_4_C_PLUS:
    if check_4cplus_version() >= 1.2:
        from adafruit_blinka.board.radxa.rockpi4cplus import *
    else:
        from adafruit_blinka.board.radxa.rockpi4cplus_legacy import *
```

In the end it really was not needed to test each pin with a LED, but I think the extra confirmation won't hurt.

makermelissa, 2023-02-28:
Hi, board detection is handled by https://github.com/adafruit/Adafruit_Python_PlatformDetect

I have a guide for that as well here: https://learn.adafruit.com/adding-a-single-board-computer-to-platformdetect-for-blinka

As for the SPI question, I'm not that familiar with the ROCK Pi boards. SPI isn't usually hooked to TX, which is for the Serial UART. Usually it's connected to SCK, MOSI, MISO, and CE. However, some board allow pins to have multiple modes, so if the TX pin can function as one of the SPI pins, then that may be the case that you would hook it up there.

I think having a legacy option like you proposed would be a good way to do it.

MathijsNL, 2023-03-01:
Thanks for linking the info about Platformdetect. Will try to do that over the weekend.

In the meantime I forked the Blinka repo and made the required changes to the board and pin file.

https://github.com/MathijsNL/Adafruit_Blinka/commits?author=MathijsNL

A few questions remain:
- Should board file aliases point to the first occurrence of a function? So if SPI1 and SPI2 are available it should point to 1?
- Can I do a PR for the board and pin file changes if the files are according to Radxa specs, or do the files require checking with hardware?
- It seems Radxa has the same pin layout and GPIO numbering for the complete Rock4 series. Is it preferred for each platform to have their own pin file, or can they share a single pin file if they are exactly the same?

I think the RK3399 platform is also using an older pin mapping but I will finish this issue first and then look into that in a separate issue.

MathijsNL, 2023-03-02:
> I usually do whatever is likely to be enabled by default.
Sounds logical, I will check it like that.

> if you could at least check some of the pins
I will try to do a simple IO test for digital outputs on all the pins with a led and test the UART. I think that should be a good starting point.

> Yes, they can share a pin file (and even a board file if applicable).
That is nice, I have a Rock4B+ and a Rock4C+ available for testing so this could go very quick once the 4C+ is fixed.

MathijsNL, 2023-03-03:
To test the GPIOs I made this streamlit app that will show the status of each GPIO
[Streamlit gist](https://gist.github.com/MathijsNL/18fb9d262b550b93a55cbac05292daae)

I am not too sure about how the Rock GPIOs should work, but I needed a pull down resistor on most pins. With all pins in INPUT mode there are a lot of pins still showing True. It seems that it is not possible to set pull up / down on the GPIOs.

Good news is that all the pins are working both as output and input. Only D26 is not working as expected yet, but it is also not documented by Radxa it seems.

Value   | Pin# L | Pin# R | Value R |  _
-- | -- | -- | -- | --
\- | D1 | D2 | \-
True | D3 | D4 | \-
True | D5 | D6 | \-
True | D7 | D8 | True
\- | D9 | D10 | True
False | D11 | D12 | True
False | D13 | D14 | \-
False | D15 | D16 | False
\- | D17 | D18 | False
True | D19 | D20 | \-
True | D21 | D22 | False
True | D23 | D24 | True
\- | D25 | D26 | \-
True | D27 | D28 | True
True | D29 | D30 | \-
True | D31 | D32 | True
True | D33 | D34 | \-
True | D35 | D36 | \-
False | D37 | D38 | True
\- | D39 | D40 | True

makermelissa, 2023-03-03:
Oh, that's a really cool way to do it. That's fine if you can't get all of the inputs working. Some of the boards just work like that and do require a pull up or pull down resistor.. At least since they are all working as outputs, you know they are mapped correctly, which was my main concern.

Gerriko, 2023-03-21:
Is it possible to expand the pin definitions to explain use for SPI, I2C and UART. I ran one of the Blinka GPIO examples and it spits this useful table out for me:

hello blinka!
Found system type: ROCK_PI_4C+ (sys.platform linux implementation cpython) 
board contents:  ['ADC_IN0', 'CS', 'D10', 'D11', 'D12', 'D13', 'D15', 'D16', 'D18', 'D19', 'D21', 'D22', 'D23', 'D24', 'D27', 'D28', 'D29', 'D3', 'D31', 'D32', 'D33', 'D35', 'D36', 'D37', 'D38', 'D40', 'D5', 'D7', 'D8', 'I2C', 'MISO', 'MOSI', 'PWM0', 'PWM1', 'SCK', 'SCL', 'SCL2', 'SCL6', 'SCL7', 'SCLK', 'SDA', 'SDA2', 'SDA6', 'SDA7', 'SPI', 'UART2_RX', 'UART2_TX', 'UART4_RX', 'UART4_TX', 'UART_RX', 'UART_TX', '__blinka__', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__repo__', '__spec__', '__version__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']

I then tried running the PN532 example for Blinka but it throws an error because of the use of board.xxxx pin definitions:

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)

Maybe we need some sort of pin definition file for the Rock4 to be able to handle these examples without error.

makermelissa, 2023-03-22:
@Gerriko make sure you have SPI enabled. Part of the issue is that Blinka was built to kind of emulate the basic CircuitPython modules and it typically only has a single I2C and SPI port (with the exception of boards like the pico where it's configurable). To easily get around this limitation, we have the Extended Bus library here: https://github.com/adafruit/Adafruit_Python_Extended_Bus. It allows you to easily use any /dev/spidexX.X devices by directly initializing them and faking anything the base system thinks it needs. It also works for I2C in the same manner.

Gerriko, 2023-04-16:
I have just discovered the reason for my SPI error noted above.
In the boards file for the Rock4 (https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/board/radxa/rockpi4.py) the pin assignments are incorrect for SPI:

SCLK should be D23 and not D19
MOSI should be D19 and not D21
MISO should be D21 and not D23

MathijsNL, 2023-04-16:
@Gerriko Thanks for spotting this error. I made a PR for it.

makermelissa, 2023-04-17:
Fixed by #674.

closed 2023-04-17 by makermelissa