adafruit/Adafruit_Blinka #657: add support in Blinka for orange pi zero 2

radfordwill, 2023-02-28:
Do you plan on supporting the orange pi zero 2? 

board info 
root@orangepizero2:~/car# cat /etc/*-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.2 LTS"
# PLEASE DO NOT EDIT THIS FILE
BOARD=orangepizero2
BOARD_NAME="Orange Pi Zero2"
BOARDFAMILY=sun50iw9
BUILD_REPOSITORY_URL=https://github.com/orangepi-xunlong/orangepi-build
BUILD_REPOSITORY_COMMIT=7a67c95-dirty
DISTRIBUTION_CODENAME=jammy
DISTRIBUTION_STATUS=supported
VERSION=3.0.6
LINUXFAMILY=sun50iw9
ARCH=arm64
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
IMAGE_UUID=32059672-6ddd-45ad-ace5-2d604fd45b70
# PLEASE DO NOT EDIT THIS FILE
BOARD=orangepizero2
BOARD_NAME="Orange Pi Zero2"
BOARDFAMILY=sun50iw9
BUILD_REPOSITORY_URL=https://github.com/orangepi-xunlong/orangepi-build
BUILD_REPOSITORY_COMMIT=7a67c95-dirty
DISTRIBUTION_CODENAME=jammy
DISTRIBUTION_STATUS=supported
VERSION=3.0.6
LINUXFAMILY=sun50iw9
ARCH=arm64
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
BRANCH=current
PRETTY_NAME="Ubuntu 22.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.2 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

makermelissa, 2023-09-13:
It appears Orange Pi Zero2 support was added in #413 and detection for the board was added in https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/141 and https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/160 as part of the 3.10.0 release of PlatformDetect. The issue is likely that you are running ubuntu and it was added with armbian.

Closing since there is an existing issue at https://github.com/adafruit/Adafruit_Python_PlatformDetect/issues/285 open for these sort of issues.

closed 2023-09-13 by makermelissa

OldSurferDude, 2023-09-13:
Thank you for the update.  I would normally confirm that the update
resolved my issue, but right now my orange pi running ubuntu is an
important tool in my development or another project.

I appreciate your efforts, as the Orange Pi can be a much less costly
solution in many instances.  Not just monetarily, but also in power usage
(~0.35A vs ~2.0A)

OSD

makermelissa, 2023-09-13:
While the issue isn't quite resolved, you may wish to follow the issue at https://github.com/adafruit/Adafruit_Python_PlatformDetect/issues/285 for further updates.