adafruit/Adafruit_Blinka #659: Pin numbers on Rock4 not matching with actual device

MathijsNL, 2023-03-04:
When working on the Rock 4C+ I noticed that the pinout for the Rock 4 was not matching the pinout on the [Radxa wiki](https://wiki.radxa.com/Rock4/hardware/gpio).

I just tested the Rock4 B+ and it seems that the pin mapping is the same. All pins were tested as output with a LED.

The quickest fix is to just copy over the content from the Rock 4C+ pin and board file to the Rock 4B+ file. I think it would be better though to just have the board.py point these 2 boards to the same files:

``` 
elif board_id == ap_board.ROCK_PI_4:
    from adafruit_blinka.board.radxa.rockpi4 import *

elif board_id == ap_board.ROCK_PI_4_C_PLUS:
    from adafruit_blinka.board.radxa.rockpi4 import * # <-- rockpi4 instead of rockpi4cplus
```

It seems the full Rock4 family is pin compatible (except for the 4C+ pre-prod version) so this would probably work. I don't have the 4C+ version that has the old layout, so if someone really needs to make that board compatible I think it can always be added again as ROCK_PI_4_C_PLUS_PREPROD if someone is willing to take the effort of doing so. 

If someone (@makermelissa  ?) could share their opinion about the best approach here I will finish the work and create a PR.

Gerriko, 2023-03-09:
Was about to try Adafruit_Blinka on a Rock4 SE board. Would I have to make a change?

MathijsNL, 2023-03-09:
The pinout as proposed in this PR #658 should work on all Rock 4 boards. So if you swap out the files it should work.

For now I am waiting till my PR gets merged, then I will fix the normal Rock Pi 4.

makermelissa, 2023-03-09:
@MathijsNL done. :)

Gerriko, 2023-03-12:
Probably a question for the Adafruit forum or for the documentation section (not sure). 
1. Is there any install and usage guidance to get the Adafruit_Blinka installed and then working on a Rock4.
2. The I am assuming that Rock4 will be using libmraa or the mraa functions. Is that the case. If so, how do you get round the root problem as I can only get mraa to work as root and not as user.

MathijsNL, 2023-03-12:
@Gerriko 
1) I was using Ubuntu 20 with kernel 4.4 and there was nothing else besides the Adafruit Blinka library that I needed to install.
2) You can fix the permissions by doing the following:
```
sudo groupadd gpio
sudo usermod -a -G gpio $USER
sudo touch /etc/udev/rules.d/99-rockchip-permissions.rules
cat <<EOF > /etc/udev/rules.d/99-rockchip-permissions.rules
# GPIO permissions
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c 'chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio; chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio'"
EOF
sudo reboot now
```

I think this fixes it, let me know if it works for you.

Gerriko, 2023-03-13:
Great, thanks for that script. I was struggling to resolve. I'm aldo using
the same combo.

makermelissa, 2023-03-13:
@Gerriko, there isn't a guide at this point because we don't have a Rock4. In fact, most of the boards added to Blinka have been by community members. However, you may be able to follow one of the other guides and it might get you pretty close.

Gerriko, 2023-03-14:
> I think this fixes it, let me know if it works for you.

@MathijsNL TL:DR It works, but need sudo

I performed the following tests. 

First, I tried using GPIO directly as per instruction on the bottom of this page: https://wiki.radxa.com/Rockpi4/hardware/gpio

Before making the fixes suggested, nothing worked. The first command "echo 157 > export" would get rejected with permission denied. After making the fixes, this command worked but then this command "echo out > direction" got rejected with a permission denied.

Second, I tried some of the commands from https://wiki.radxa.com/Rockpi4/dev/libmraa
Here, if I used "mraa-gpio set 40 1  //pin40 pull high " nothing happens.
If I use "sudo mraa-gpio set 40 1  //pin40 pull high" it works.

Thus to get my simple GPIO Python sketches to work I needed to use "sudo Python3 PythonSketchName".
Not sure if that is intended behaviour.

I tried a bit of online searching and came across this article - not sure if it is relevant here: https://askubuntu.com/questions/1230947/gpio-for-raspberry-pi-gpio-group

MathijsNL, 2023-03-14:
I will have a look at it, maybe I forgot one important step here.

Gerriko, 2023-03-14:
I tested spi on rock4 using mraa spidev without sudo and it returned errors. 
I found this article which seems to have collated everything I have found in various articles quite nicely. Hopefully it helps. https://forum.up-community.org/discussion/2141/solved-tutorial-gpio-i2c-spi-access-without-root-permissions

MathijsNL, 2023-03-15:
@Gerriko I updated my comment above with the proper udev rules. Those should work.

closed 2023-03-16 by makermelissa