adafruit/Adafruit_Blinka #660: MCP2221 pins driven low during import of board

Rainerlan, 2023-03-05:
I try to control shutters. Their movement is being triggered by a LOW input signal.
Hence, the shutter-input MUST stay high (unless commanded to move otherwise).
I connected G0 (later configured as OUTPUT) to the shutter input pin.

Without a pullup at G0, when executing "import board" in python3, G0 pin is being pulled to ground several times.
After attaching a pullup (100kOhm) at G0, still after executing "import board", G0 pin is being pulled to ground for ~12ms.

Note: This is before digitalio is being imported and before the pins are being setup as OUTPUT.
How can I avoid G0 being pulled to low during startup of the script (import of board)?

caternuson, 2023-03-06:
Try suppressing the software reset by setting the environment variable `MCP2221_RESET_DELAY` to `0`. Exactly how to do that is OS dependent. But it'd be the same way you are setting `BLINKA_MCP2221` to `1`.

```

caternuson, 2023-03-06:
Although the pins are also set to inputs at startup:
https://github.com/adafruit/Adafruit_Blinka/blob/802e6874ffa75449f61e229e7aeb00fc5e0022ab/src/adafruit_blinka/microcontroller/mcp2221/mcp2221.py#L64-L66

Rainerlan, 2023-03-06:
`export MCP2221_RESET_DELAY=0 ` seems to improve the situation (don't know why, because in the code in checks for `>=0 MCP2221_RESET_DELAY`) - but I still get a 2ms pulse to ground (when pulling up the pin using a 1kOhms resistor). Although the 2ms puls seems to be to short to be recognized by the Velux remote, its still kind of annoying.

The other problem is, that when setting the pins to output, the GPIO output is set to low/GND (False).
Maybe the GPx Power-up Settings (GPIO Output value) in EEPROM are set to "0"?

Unfortunately the read flash data command is not yet supported in MCP2221.py - So I can't read it out - and the factory default is not stated in the datasheet.

Rainerlan, 2023-03-06:
After applying the above changes, the GPIOs work as I need them to work (default set to HIGH/3.3V when changing them to outputs - also the reset is not messing around with the pins output states)

caternuson, 2023-03-06:
Oops, sry, meant negative number for the reset envar.

Good hacks for the other parts! Glad you got it working.

Rainerlan, 2023-03-06:
@caternuson Thanks a lot for your support!