adafruit/Adafruit_Blinka #666: Orange pi 5 RK3388 no support?

drozdi70, 2023-03-24:
Hi

Issue with orange pi 5 board:

orangepi@orangepi5:/usr/local/lib/python3.9/dist-packages$ cat /etc/*-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=orangepi5
BOARD_NAME="Orange Pi 5"
BOARDFAMILY=rockchip-rk3588
BUILD_REPOSITORY_URL=https://github.com/orangepi-xunlong/orangepi-build
BUILD_REPOSITORY_COMMIT=6a1ff9f48-dirty
DISTRIBUTION_CODENAME=bullseye
DISTRIBUTION_STATUS=supported
VERSION=1.1.4
LINUXFAMILY=rockchip-rk3588
ARCH=arm64
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
IMAGE_UUID=c1f9cfc0-b96e-421b-b24c-fd3bda32d507
# PLEASE DO NOT EDIT THIS FILE
BOARD=orangepi5
BOARD_NAME="Orange Pi 5"
BOARDFAMILY=rockchip-rk3588
BUILD_REPOSITORY_URL=https://github.com/orangepi-xunlong/orangepi-build
BUILD_REPOSITORY_COMMIT=6a1ff9f48-dirty
DISTRIBUTION_CODENAME=bullseye
DISTRIBUTION_STATUS=supported
VERSION=1.1.4
LINUXFAMILY=rockchip-rk3588
ARCH=arm64
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
BRANCH=legacy
PRETTY_NAME="Orange Pi 1.1.4 Bullseye"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

orangepi@orangepi5:~/work$ python3 board_detect.py
Chip id:  RK3588
Board id:  None
Pi 3B+?  False
BBB?  False
Orange Pi PC?  False
generic Linux PC?  False

BLINKA doesn't support orange PI 5??

If changed in board.py line 130 into 
elif detector.chip.id == 'RK3588':
 from adafruit_blinka.board.orangepi.orangepi5 import *

orangepi@orangepi5:/usr/local/lib/python3.9/dist-packages$ python3 board.py
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/board.py", line 131, in <module>
    from adafruit_blinka.board.orangepi.orangepi5 import *
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/board/orangepi/orangepi5.py", line 6, in <module>
    from adafruit_blinka.microcontroller.rockchip.rk3588 import pin
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/rockchip/rk3588/pin.py", line 11, in <module>
    GPIO0_A0 = Pin((0, 0))
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 35, in __init__
    self._chip = gpiod.Chip(str(pin_id[0]), gpiod.Chip.OPEN_BY_NUMBER)
AttributeError: type object 'Chip' has no attribute 'OPEN_BY_NUMBER'

Kindly
Robert

drozdi70, 2023-03-26:
I have also checked on Armbian 

/etc/armbian-release
::::::::::::::
# PLEASE DO NOT EDIT THIS FILE
BOARD=orangepi5
BOARD_NAME="Orange Pi 5"
BOARDFAMILY=rockchip-rk3588
BUILD_REPOSITORY_URL=https://github.com/armbian/build
BUILD_REPOSITORY_COMMIT=1a8daf0
VERSION=23.02.2
LINUXFAMILY=rockchip-rk3588
ARCH=arm64
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=stable
BRANCH=legacy
::::::::::::::
/etc/lsb-release
::::::::::::::
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.2 LTS"
::::::::::::::
/etc/os-release
::::::::::::::
PRETTY_NAME="Ubuntu 22.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.2 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

and result is the same:
 File "/home/orangepi/.local/lib/python3.10/site-packages/board.py", line 131, in <module>
    from adafruit_blinka.board.orangepi.orangepi5 import *
  File "/home/orangepi/.local/lib/python3.10/site-packages/adafruit_blinka/board/orangepi/orangepi5.py", line 6, in <module>
    from adafruit_blinka.microcontroller.rockchip.rk3588 import pin
  File "/home/orangepi/.local/lib/python3.10/site-packages/adafruit_blinka/microcontroller/rockchip/rk3588/pin.py", line 11, in <module>
    GPIO0_A0 = Pin((0, 0))
  File "/home/orangepi/.local/lib/python3.10/site-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 37, in __init__
    self._chip = gpiod.chip(str(pin_id[0]), gpiod.chip.OPEN_BY_NUMBER)
  File "/home/orangepi/.local/lib/python3.10/site-packages/gpiod/libgpiodcxx/__init__.py", line 107, in __init__
    self.open(device, how)
  File "/home/orangepi/.local/lib/python3.10/site-packages/gpiod/libgpiodcxx/__init__.py", line 136, in open
    chip_struct = func(device)
  File "/home/orangepi/.local/lib/python3.10/site-packages/gpiod/libgpiod/__init__.py", line 1085, in gpiod_chip_open_by_number
    return gpiod_chip_open("/dev/gpiochip" + str(num))
  File "/home/orangepi/.local/lib/python3.10/site-packages/gpiod/libgpiod/__init__.py", line 112, in gpiod_chip_open
    fd = os_open(path, O_RDWR | O_CLOEXEC)
PermissionError: [Errno 13] Permission denied: '/dev/gpiochip0'

makermelissa, 2023-03-27:
> PermissionError: [Errno 13] Permission denied: '/dev/gpiochip0'

Did you try running with `sudo`?

drozdi70, 2023-03-28:
Hi MelissaA litle better but still some ***@***.***:~/work$ sudo python3 blinka_test.pyHello blinka!Traceback (most recent call last):  File "/home/orangepi/work/blinka_test.py", line 8, in &lt;module&gt;    pin = digitalio.DigitalInOut(board.PA6)AttributeError: module 'board' has no attribute ***@***.******@***.***:~/work$ more blinka_test.pyimport boardimport digitalioimport busioprint("Hello blinka!")# Try to great a Digital inputpin = digitalio.DigitalInOut(board.PA6)print("Digital IO ok!")# Try to create an I2C devicei2c = busio.I2C(board.SCL, board.SDA)print("I2C ok!")# Try to create an SPI device#spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)print("SPI ok!")print("done!")KindlyRobertTemat: Re: [adafruit/Adafruit_Blinka] Orange pi 5 RK3388 no support? (Issue #666)Data: 2023-03-27 19:33Nadawca: "Melissa LeBlanc-Williams" ***@***.***&gt;Adresat: "adafruit/Adafruit_Blinka" ***@***.***&gt;; DW: "drozdi70" ***@***.***&gt;; "Author" ***@***.***&gt;; PermissionError: [Errno 13] Permission denied: '/dev/gpiochip0'Did you try running with sudo?

makermelissa, 2023-03-28:
That's a bit garbled, but I think you were trying to say:
> Hi Melissa
>
> A litle better but still some
> ```
> Hello blinka!
> Traceback (most recent call last):
> File "/home/orangepi/work/blinka_test.py", line 8, in <module>
> pin = digitalio.DigitalInOut(board.PA6)
> AttributeError: module 'board' has no attribute
> 
> import board
> import digitalio
> import busio
> print("Hello blinka!")# Try to great a Digital input
> pin = digitalio.DigitalInOut(board.PA6)
> print("Digital IO ok!")# Try to create an I2C device
> i2c = busio.I2C(board.SCL, board.SDA)
> print("I2C ok!")# Try to create an SPI device
> #spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
> print("SPI ok!"
> print("done!")
> ```
> Kindly
> Robert

For future replies, we would appreciate replying on GitHub because email is notoriously bad at scrambling GitHub messages. It looks like you are trying to use board.PA6. However, according to the board definition at https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/board/orangepi/orangepi5.py, PA6 isn't a valid name. It appears they all begin with `D`.

drozdi70, 2023-03-28:
HiSorry but it is mail garbage putting all together...Im referring to official pages:https://learn.adafruit.com/circuitpython-on-orangepi-linux/orange-pi-pc-setuphttps://learn.adafruit.com/circuitpython-on-orangepi-linux/orange-pi-r1and there in both case blinkatest.py points to pin = digitalio.DigitalInOut(board.PA6)how should it be for orange pi 5 as not presented?KindlyRobertTemat: Re: [adafruit/Adafruit_Blinka] Orange pi 5 RK3388 no support? (Issue #666)Data: 2023-03-28 17:21Nadawca: "Melissa LeBlanc-Williams" ***@***.***&gt;Adresat: "adafruit/Adafruit_Blinka" ***@***.***&gt;; DW: "drozdi70" ***@***.***&gt;; "Author" ***@***.***&gt;; 

Hi MelissaA litle better but still some @.:~/work$ sudo python3 blinka_test.pyHello blinka!Traceback (most recent call last):  File "/home/orangepi/work/blinka_test.py", line 8, in &lt;module&gt;    pin = digitalio.DigitalInOut(board.PA6)AttributeError: module 'board' has no attribute @.@.:~/work$ more blinka_test.pyimport boardimport digitalioimport busioprint("Hello blinka!")# Try to great a Digital inputpin = digitalio.DigitalInOut(board.PA6)print("Digital IO ok!")# Try to create an I2C devicei2c = busio.I2C(board.SCL, board.SDA)print("I2C ok!")# Try to create an SPI device#spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)print("SPI ok!")print("done!")KindlyRobertTemat: Re: [adafruit/Adafruit_Blinka] Orange pi 5 RK3388 no support? (Issue #666)Data: 2023-03-27 19:33Nadawca: "Melissa LeBlanc-Williams" @.&gt;Adresat: "adafruit/Adafruit_Blinka" @.&gt;; DW: "drozdi70" @.&gt;; "Author" @.&gt;; PermissionError: [Errno 13] Permission denied: '/dev/gpiochip0'Did you try running with sudo?

That's a bit garbled, but I think you were trying to say:

Hi Melissa
A litle better but still some
Hello blinka!
Traceback (most recent call last):
File "/home/orangepi/work/blinka_test.py", line 8, in &lt;module&gt;
pin = digitalio.DigitalInOut(board.PA6)
AttributeError: module 'board' has no attribute

import board
import digitalio
import busio
print("Hello blinka!")# Try to great a Digital input
pin = digitalio.DigitalInOut(board.PA6)
print("Digital IO ok!")# Try to create an I2C device
i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")# Try to create an SPI device
#spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
print("SPI ok!"
print("done!")

Kindly
Robert

For future replies, we would appreciate replying on GitHub because email is notoriously bad at scrambling GitHub messages. It looks like you are trying to use board.PA6. However, according to the board definition at https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/board/orangepi/orangepi5.py, PA6 isn't a valid name. It appears they all begin with D.

anguselheimStudios, 2023-04-22:
Hi Robert, Looking at the definition, they are numbered left to right top to bottom with the format D#.

**So you would have :**
```
3.3V - 5V 
  D3 - 5V
  D5 - GND
  D7 - D8
 GND - D10
```
Etc...

I don't have that board, so I can't confirm for you. You can open up the REPL, import board, and run dir(board).  It will tell you what pins you have available to you.

This is what it looks like on my aml-s905x-cc:
```
>>> import board
>>> dir(board)
['I2C', 'MISO', 'MOSI', 'P10', 'P12', 'P13', 'P15', 'P16', 'P18', 
'P19', 'P21', 'P22', 'P23', 'P24', 'P26', 'P27', 'P28', 'P29', 
'P2J12', 'P2J13', 'P2J33', 'P2J34', 'P2J35', 'P2J36', 'P3', 'P31', 
'P32', 'P33', 'P35', 'P36', 'P37', 'P38', 'P40', 'P5', 'P7', 'P8', 
'P9J12', 'SCL', 'SCL0', 'SCL1', 'SCLK', 'SDA', 'SDA0', 'SDA1', 
'SPI', 'SPI_CS', '__blinka__', '__builtins__', '__cached__', 
'__doc__', '__file__', '__loader__', '__name__', '__package__', 
'__repo__', '__spec__', '__version__', 'ap_board', 'board_id', 
'detector', 'it', 'pin', 'sys']

```

This is one of the few things that drives me nuts about Blinka. These boards all have pretty much the same headers; I wish we could stick to one naming convention all across the boards. Pun intended. 

Couldn't have a more fitting ticket number though.

drozdi70, 2023-04-23:
HiI have solved anf fixed some issue with your package.I attach file -&gt; orangepi5.txtAlso how it is used I put on the forum: https://forum.openmarine.net/showthread.php?tid=4559KindlyRobert Temat: Re: [adafruit/Adafruit_Blinka] Orange pi 5 RK3388 no support? (Issue #666)Data: 2023-04-23 0:35Nadawca: "anguselheimStudios" ***@***.***&gt;Adresat: "adafruit/Adafruit_Blinka" ***@***.***&gt;; DW: "drozdi70" ***@***.***&gt;; "Author" ***@***.***&gt;; 

Hi Robert, Looking at the definition, they are numbered left to right top to bottom with the format D#.
So you would have :
3.3V - 5V 
  D3 - 5V
  D5 - GND
  D7 - D8
 GND - D10

Etc...
I don't have that board, so I can't confirm for you. You can open up the REPL, import board, and run dir(board).  It will tell you what pins you have available to you.
This is what it looks like on my aml-s905x-cc:

>>> import board
>>> dir(board)
['I2C', 'MISO', 'MOSI', 'P10', 'P12', 'P13', 'P15', 'P16', 'P18', 
'P19', 'P21', 'P22', 'P23', 'P24', 'P26', 'P27', 'P28', 'P29', 
'P2J12', 'P2J13', 'P2J33', 'P2J34', 'P2J35', 'P2J36', 'P3', 'P31', 
'P32', 'P33', 'P35', 'P36', 'P37', 'P38', 'P40', 'P5', 'P7', 'P8', 
'P9J12', 'SCL', 'SCL0', 'SCL1', 'SCLK', 'SDA', 'SDA0', 'SDA1', 
'SPI', 'SPI_CS', '__blinka__', '__builtins__', '__cached__', 
'__doc__', '__file__', '__loader__', '__name__', '__package__', 
'__repo__', '__spec__', '__version__', 'ap_board', 'board_id', 
'detector', 'it', 'pin', 'sys']

This is one of the few things that drives me nuts about Blinka. These boards all have pretty much the same headers; I wish we could stick to one naming convention all across the boards. Pun intended.
Couldn't have a more fitting ticket number though.

Python 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> dir(board)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'board' is not defined
>>> import board
>>> dir (board)
['CS', 'D10', 'D11', 'D12', 'D13', 'D15', 'D16', 'D18', 'D19', 'D21', 'D22', 'D23', 'D24', 'D26', 'D3', 'D5', 'D7', 'D8', 'I2C', 'I2C1_SCL', 'I2C1_SDA', 'I2C3_SCL', 'I2C3_SDA', 'I2C5_SCL', 'I2C5_SDA', 'MISO', 'MOSI', 'RX', 'RXD', 'SCL', 'SCLK', 'SDA', 'SPI', 'SPI4_CLK', 'SPI4_CS1', 'SPI4_MISO', 'SPI4_MOSI', 'TX', 'TXD', 'UART0_RX', 'UART0_TX', 'UART1_RX', 'UART1_TX', 'UART3_RX', 'UART3_TX', 'UART4_RX', 'UART4_TX', '__blinka__', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__repo__', '__spec__', '__version__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']

also
 +------+-----+----------+--------+---+   OPI5   +---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 |      |     |     3.3V |        |   |  1 || 2  |   |        | 5V       |     |      |
 |   47 |   0 |    SDA.5 |   ALT9 | 1 |  3 || 4  |   |        | 5V       |     |      |
 |   46 |   1 |    SCL.5 |   ALT9 | 1 |  5 || 6  |   |        | GND      |     |      |
 |   54 |   2 |    PWM15 |     IN | 1 |  7 || 8  | 1 | ALT10  | RXD.0    | 3   | 131  |
 |      |     |      GND |        |   |  9 || 10 | 1 | ALT10  | TXD.0    | 4   | 132  |
 |  138 |   5 |  CAN1_RX |     IN | 1 | 11 || 12 | 1 | IN     | CAN2_TX  | 6   | 29   |
 |  139 |   7 |  CAN1_TX |     IN | 1 | 13 || 14 |   |        | GND      |     |      |
 |   28 |   8 |  CAN2_RX |     IN | 1 | 15 || 16 | 1 | ALT10  | SDA.1    | 9   | 59   |
 |      |     |     3.3V |        |   | 17 || 18 | 1 | ALT10  | SCL.1    | 10  | 58   |
 |   49 |  11 | SPI4_TXD |     IN | 1 | 19 || 20 |   |        | GND      |     |      |
 |   48 |  12 | SPI4_RXD |     IN | 1 | 21 || 22 | 1 | IN     | GPIO2_D4 | 13  | 92   |
 |   50 |  14 | SPI4_CLK |     IN | 1 | 23 || 24 | 1 | IN     | SPI4_CS1 | 15  | 52   |
 |      |     |      GND |        |   | 25 || 26 | 1 | IN     | PWM1     | 16  | 35   |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+   OPI5   +---+--------+----------+-----+------+

import board
import digitalio
import busio

print("Hello blinka!")

# Try to great a Digital input
pin = digitalio.DigitalInOut(board.D10)
print("Digital IO ok!")

# Try to create an I2C device
i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")

# Try to create an SPI device
#spi = busio.SPI(board.SCLK, board.MOSI, board.MISO) --not needed now
print("SPI N/A!")

print("done!")

Hello blinka!
Digital IO ok!
I2C ok!
SPI N/A!
done!

drozdi70, 2023-12-18:
I suggest to: 

1. In board.py in /usr/local/lib/python3.11/dist-packages:

elif board_id == ap_board.ORANGE_PI_5:
    from adafruit_blinka.board.orangepi.orangepi5 import *

elif board_id == ap_board.ORANGE_PI_5_PLUS:
    from adafruit_blinka.board.orangepi.orangepi5plus import *

already done :)

2. Orangepi 5 support (rockchip 3588S):
in /usr/local/lib/python3.11/dist-packages/adafruit_blinka/board/orangepi/orangepi5.py

use from adafruit_blinka.microcontroller.rockchip.rk3588s import 

in /usr/local/lib/python3.11/dist-packages/adafruit_blinka/microcontroller/rockchip/rk3588s place pin.py related to Orangepi 5

3. Orangepi 5 plus support (rockchip 3588):
in /usr/local/lib/python3.11/dist-packages/adafruit_blinka/board/orangepi/orangepi5plus.py

use from adafruit_blinka.microcontroller.rockchip.rk3588 import pin

in /usr/local/lib/python3.11/dist-packages/adafruit_blinka/microcontroller/rockchip/rk3588 place pin.py related Orangepi 5 Plus