adafruit/Adafruit_Blinka #675: "IndexError: tuple index out of range" importing board in Armbian on Odroid N2+

anguselheimStudios, 2023-04-16:
*This is my firs time reporting an issue, so please let me know if I'm sharing to much or little, or whatever other faux pas I may be making.*

Running python without sudo I get a "permission denied" message when I try to import board.

Running python as sudo I get this:

```
angus@odroidn2:~$ sudo python3
[sudo] password for angus: 
Python 3.10.6 (main, Mar 10 2023, 10:55:28) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.10/dist-packages/board.py", line 188, in <module>
    from adafruit_blinka.board.hardkernel.odroidn2 import *
  File "/usr/local/lib/python3.10/dist-packages/adafruit_blinka/board/hardkernel/odroidn2.py", line 62, in <module>
    SCL = pin.i2cPorts[1][1]
IndexError: tuple index out of range
>>> 
```
My system info:
```
angus@odroidn2:~$ neofetch
                                 angus@odroidn2 
                                 -------------- 
      █ █ █ █ █ █ █ █ █ █ █      OS: Armbian (23.02.2) aarch64 
     ███████████████████████     Host: Hardkernel ODROID-N2Plus 
   ▄▄██                   ██▄▄   Kernel: 6.1.11-meson64 
   ▄▄██    ███████████    ██▄▄   Uptime: 32 mins 
   ▄▄██   ██         ██   ██▄▄   Packages: 1592 (dpkg) 
   ▄▄██   ██         ██   ██▄▄   Shell: bash 5.1.16 
   ▄▄██   ██         ██   ██▄▄   Resolution: 1680x1050 
   ▄▄██   █████████████   ██▄▄   DE: Cinnamon 5.2.7 
   ▄▄██   ██         ██   ██▄▄   WM: Mutter (Muffin) 
   ▄▄██   ██         ██   ██▄▄   WM Theme: (Numix) 
   ▄▄██   ██         ██   ██▄▄   Theme: Numix [GTK2/3] 
   ▄▄██                   ██▄▄   Icons: Numix [GTK2/3] 
     ███████████████████████     Terminal: gnome-terminal 
      █ █ █ █ █ █ █ █ █ █ █      CPU: ARMv8 rev 2 (v8l) (6) @ 2.016GHz 
                                 Memory: 1986MiB / 3785MiB 

angus@odroidn2:~$ apt list | grep python3-libgpiod

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

python3-libgpiod/jammy,now 1.6.3-1build1 arm64 [installed]
python3-libgpiod/jammy 1.6.3-1build1 armhf
angus@odroidn2:~$ pip3 list | grep Adafruit
Adafruit-Blinka                  8.18.0
Adafruit-PlatformDetect          3.45.1
Adafruit-PureIO                  1.1.10
angus@odroidn2:~$ pip3 list | grep circuit
adafruit-circuitpython-busdevice 5.2.4
adafruit-circuitpython-requests  1.13.1
adafruit-circuitpython-typing    1.9.1
```

I went through the same install steps as I did with my Libre Computer AML-S905X-CC "Le Potato", which worked successfully. 

I have also successfully run a simple blink script using gpiod directly in python3, so I'm pretty sure libgpiod is OK.

makermelissa, 2023-04-17:
It looks like it was implemented in a very confusing way, but apparently necessary due to various versions of the same board.

If you type `ls /dev/i2c*`, what do you see? It appears there was no error handling added for the section, so it's expecting 2 I2C ports and not seeing that.

anguselheimStudios, 2023-04-17:
There are two ports:

```
angus@odroidn2:~$ ls /dev/i2c*
/dev/i2c-0  /dev/i2c-1
```

Yeah, Hardkernel kind of sort of treat the n2 and the n2+ as the same thing kind of maybe a little, but not always. It's a little strange.

makermelissa, 2023-09-08:
I think a good solution will be to add a check to see if the structure of i2cPorts matches what the code expects.

PR #705 (2023-09-11): Add check to ODroid n2 to make sure i2c matches expected ports
Fixes #675.
closed 2023-09-11 by makermelissa