adafruit/Adafruit_Blinka #676: Blinka referencing sysfs_pin.py and not libgpiod_pin.py

Gerriko, 2023-04-18:
I am using a Rock4 C+ board and I am developing my own SPI library.

For some reason I am getting an error:
`adafruit_blinka.microcontroller.generic_linux.sysfs_pin.GPIOError: [Errno 22] Unexporting GPIO: Invalid argument`

What I cannot work out (as in I don't know where to look) is why is Blinka using sysfs_pin.py when it should really be using libgpiod_pin.py

Gerriko, 2023-04-18:
In order to use libgpiod library with the rk3399, I discovered that the GPIO pin definitions found in the rk3588 pin.py file are perfectly suited here too (as a replacement): 
https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/microcontroller/rockchip/rk3588/pin.py

Gerriko, 2023-04-23:
@makermelissa I am wondering if there is a way to handle both sysfs_pin and libgpiod_pin inside the microcontroller pin.py file.

I have a similar issue with the Rock CM3 board, but this time in reverse, where the official (and working) Radxa distro release is still using sysfs_pin. However, most other boards, which also use the RK3566 chip, will be using libgpiod_pin.

Can PlatformDetect be used here?

What other options are possible to allow for both use cases?

makermelissa, 2023-05-01:
Perhaps you have have 2 versions of the RK3566 with it specified as libgpiod or sysfs. So maybe it would be something like `rk3566_sysfs/` and `rk3566_libgpiod/`. Then you can point the board to the appropriate one.

makermelissa, 2023-05-01:
Or better yet, maybe create `sysfs` and `libgpiod` folders inside of the `rk3566` folder.

makermelissa, 2023-09-13:
Fixed via #677.

closed 2023-09-13 by makermelissa