adafruit/Adafruit_Blinka #680: NotImplementedError: Board not supported TI_J721E_SK

ss13ms110, 2023-04-28:
Hello Devs,

I am trying to use BNO055 breakout board with BeagleBone AI-64.

After PyPI installation of `adafruit-circuitpython-bno055`, when I `import board` I get this error.

```
debian@BeagleBone:~/test/i2c$ python 
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import adafruit_bno055
beagle,j721e-beagleboneai64ti,j721e
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/board.py", line 361, in <module>
    raise NotImplementedError(f"Board not supported {board_id}.")
NotImplementedError: Board not supported TI_J721E_SK.
```

Is BB-AI-64 not supported yet? if not, is there any workaround that can be done to make it work.

Please help.

Thank you
Best
Shubham

makermelissa, 2023-05-01:
I don't believe this was added yet. I just did a search and it looks like I missed the announcement of this one. Want to submit a PR? You might be able to use a lot of the work done for the original BeagleBone AI.

ss13ms110, 2023-05-02:
Hello Melissa,

I have made changes in the code to support BeagleBone-AI-64. Not only just `adafruit_blinka`, changes were needed in `microcontroller` and `adafruit_platformdetect` to support BB-AI-64. Let me know how can I share those changes as a contribution to this library.

Thank you

hamzaish, 2023-05-21:
Hi, I'm currently struggling with this same error. Have you posted the solution anywhere @ss13ms110 @makermelissa ? Thanks!