adafruit/Adafruit_Blinka #682: Why can't I2C baudrate/frequency be set in Python?

erichiggins, 2023-05-02:
For all microcontrollers supported by this project, there are lines commented out that state that the I2C frequency cannot be set in Python and is therefore ignored. because these print statements are commented out, the end user (me) has no idea why the API parameter isn't working.
https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/microcontroller/generic_linux/i2c.py#L23-L24

I'm using a Raspberry Pi Zero W with the Adafruit motor control bonnet to control a stepper motor which is turning slower than I like. The Adafruit MotorKit documentation suggests updating the frequency to get more speed.
https://docs.circuitpython.org/projects/motorkit/en/latest/api.html#adafruit_motorkit.MotorKit

I've already changed the `config.txt` setting to update the I2C speed from 100khz to 400khz, but my understanding is that the code also needs to use it. I've confirmed that it's now 400khz.
https://learn.adafruit.com/raspberry-pi-i2c-clock-stretching-fixes/change-the-clock-speed

Three questions:

1. Why is this setting ignored, or, why can't it be changed in Python?
2. Can these lines be uncommented and changed to throw a Warning instead of ignoring the parameter silently?
3. If I can't changed the speed in Python, what should I do instead?

caternuson, 2023-05-02:
1. It's configured at boot time.
2. Throwing an exception instead of silently ignoring might be a better option.
3. By editing the config.txt file. [see here for an example](https://learn.adafruit.com/raspberry-pi-i2c-clock-stretching-fixes/change-the-clock-speed)

erichiggins, 2023-05-02:
@caternuson Thank you! Would you like me to create a PR to throw a warning? Are the docs maintained in this repo as well?

ladyada, 2023-05-03:
i believe it used to throw an exception but some libraries don't catch the exception - so perhaps a print statement would be more useful

anguselheimStudios, 2023-05-03:
Seems like that should be in the docstring too. Same with the phony pins in busio. Would have saved me some time troubleshooting if I knew the pins I kept switching around were bogus.

PR #708 (2023-09-14): Display a message if user tries to set freq on linux
Fixes #682. Also improves issue template messages.