adafruit/Adafruit_Blinka #684: Trinkey QT 2040 U2IF, module hid device has no attribute

dapinitial, 2023-05-29:
`Traceback (most recent call last):
  File "/Users/dpuerto/Sites/xxxx.com/sensors/trinkey.py", line 2, in <module>
    import board
  File "/Users/dpuerto/Sites/xxxx.com/sensors/board/__init__.py", line 17, in <module>
    pin.GP0._u2if_open_hid(0x239A, 0x0109)
  File "/Users/dpuerto/Library/Python/3.9/lib/python/site-packages/adafruit_blinka/microcontroller/rp2040_u2if/pin.py", line 29, in _u2if_open_hid
    rp2040_u2if.open(vid, pid)
  File "/Users/dpuerto/Library/Python/3.9/lib/python/site-packages/adafruit_blinka/microcontroller/rp2040_u2if/rp2040_u2if.py", line 119, in open
    self._hid = hid.device()
AttributeError: module 'hid' has no attribute 'device'`

Hi, I've been trying to setup a new Trinkey QT2040 with an SCD41 for about 2 weeks. I can't seem to get the U2IF firmware to work. I have been able to test various features and functionality but it seems `hid` is the continuous failpoint. Maybe with a version mismatch with the adafruit-blinka package?

I am using Python 3.8 on MacOSX 13.4 Ventura and some semblance of board or busio to instantiate the I2C. I've been able to blink the LED light, etc, and run through the sanity tests for the most part. Really at a loss for what I am doing wrong. I've deleted Python and everything related to it via Homebrew and re-installed everything a couple times as well and I always arrive at the same error. I was also trying this using the CircuitPython Firmware, but running into the same error. Ultimately the U2IF is better for my use case any ways (reading the sensor data and then sending that to the cloud using signalR/websockets to then be called from my web app).

More information can be referenced here:
https://forums.adafruit.com/viewtopic.php?p=973995#p973995

caternuson, 2023-05-30:
You've probably installed more than one "hid" module. Make sure you've installed **hidapi**, not hid. There are two different modules. It's confusing, since the module is named hidapi, but the import in python code is called hid.

You can check using:
```
pip3 list
```
and search for any modules with "hid" in the name. Can probably use grep on the mac:
```
pip3 list | grep hid
```

If the module **hid** shows up in the list, **uninstall it**.

dapinitial, 2023-05-30:
Hi, thanks for getting back to me,

Yes, I have seen that link and looked into the github repo.

I agree, and you were right, I did install hid in my frustration to try and
get the dependencies to work. I also have pyhidapi from another tutorial I
was following on the website. Should I remove that as well? I have libusb,
pyserial, Adafruit-Blinka 8.19.0, board 1.0, pyusb...  I probably should
start removing everything and starting over again.

Currently: I have the environment variable set, however, when I try to
import board:

Traceback (most recent call last):

  File "/Users/dpuerto/Sites/XXXX.com/sensors/trinkey.py", line 2, in
<module>

    import board

  File
"/Users/dpuerto/Library/Python/3.9/lib/python/site-packages/board.py", line
22, in <module>

    from adafruit_blinka.agnostic import board_id, detector

  File
"/Users/dpuerto/Library/Python/3.9/lib/python/site-packages/adafruit_blinka/agnostic/__init__.py",
line 21, in <module>

    chip_id = detector.chip.id

  File
"/Users/dpuerto/Library/Python/3.9/lib/python/site-packages/adafruit_platformdetect/chip.py",
line 419, in __getattr__

    if self.id == attr:

  File
"/Users/dpuerto/Library/Python/3.9/lib/python/site-packages/adafruit_platformdetect/chip.py",
line 419, in __getattr__

    if self.id == attr:

  File
"/Users/dpuerto/Library/Python/3.9/lib/python/site-packages/adafruit_platformdetect/chip.py",
line 419, in __getattr__

    if self.id == attr:

  [Previous line repeated 978 more times]

  File
"/Users/dpuerto/Library/Python/3.9/lib/python/site-packages/adafruit_platformdetect/chip.py",
line 65, in id

    if os.environ.get("BLINKA_FT232H"):

  File
"/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/_collections_abc.py",
line 764, in get

    return self[key]

  File
"/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/os.py",
line 676, in __getitem__

    value = self._data[self.encodekey(key)]

  File
"/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/os.py",
line 755, in encode

    if not isinstance(value, str):

RecursionError: maximum recursion depth exceeded while calling a Python
object

I uninstalled and reinstalled adafruit-blinka and adafruit-platformdetect
afterwards and restarted terminal and made sure that the environment
variable returned 1 and then ran again and received the same error.

Any recommendations? I know this is going to be really cool and a great
learning tool once I overcome this hurdle, but getting the sensors to
output the data for me has proven a difficult task...

caternuson, 2023-05-30:
I think it's best to continue this in the forum thread linked above. This is most likely an issue with the current state of your Python environment. It's not an issue with the Blinka library, which is what opening an issue in this repo is for.

dapinitial, 2023-05-30:
Ok, thanks.

caternuson, 2023-05-31:
Resolved in forums:
https://forums.adafruit.com/viewtopic.php?p=974185#p974185

Closing here.

closed 2023-05-31 by caternuson