adafruit/Adafruit_Blinka #686: ImportError: no module named 'typing' pi pico

Nicchessla, 2023-06-21:
I am trying to run adafruits HX8357 display with the raspberry pico, running micropython using Blinka.
when I import displayio I get the error above

tekktrik, 2023-06-23:
Can you post the error traceback?

corlovsky, 2023-07-10:
I'm having the same issue. Please see below:

`MPY: soft reboot
Chip id:  RP2040
Board id:  RASPBERRY_PI_PICO
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
  File "/lib/displayio/__init__.py", line 20, in <module>
ImportError: no module named 'typing'
>>> `

tekktrik, 2023-07-11:
@makermelissa should the Pico be able to support importing of `typing` like the Pi proper does?  Here's the relevant line:  https://github.com/adafruit/Adafruit_Blinka_Displayio/blob/61102c93bfb90fbccc4bc21122444531ebe2dd84/displayio/__init__.py#L20

makermelissa, 2023-07-12:
@tekktrik I'm not sure, but if there is an option of not requiring it, that would be ideal since there's already so little extra space.

tekktrik, 2023-07-14:
I'm more surprised because I don't think `typing` exists and I'm surprised it might only be an issue now.  I can do some more digging.

Neradoc, 2023-07-15:
Is displayio even compatible with Micropython ? It has seemingly never been tested there.
It's not an issue to use typing without guards in a library that is not supported on MP.
This should probably be an issue in https://github.com/adafruit/Adafruit_Blinka_Displayio