adafruit/Adafruit_Blinka #689: the pin.py for allwinner-h616 has error

sc-bin, 2023-07-13:
hello，I foud a error in file
```
Adafruit_Blinka/src/adafruit_blinka/microcontroller/allwinner/h616/pin.py
```
every pins are marked as gpiochip1，Actually, they both belong to gpiochip0

I am trying to run blinka in my allwinner-h616 board，and when I run the code `led = digitalio.DigitalInOut(board.PC8)`, got the error like this
```
Traceback (most recent call last):
  File "/mnt/gpioio.py", line 7, in <module>
    led = digitalio.DigitalInOut(board.PC8)
  File "/usr/local/lib/python3.9/dist-packages/digitalio.py", line 165, in __init__
    self.direction = Direction.INPUT
  File "/usr/local/lib/python3.9/dist-packages/digitalio.py", line 195, in direction
    self._pin.init(mode=Pin.IN)
  File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 55, in init
    self._line = self._chip.get_line(int(self._num))
OSError: [Errno 22] Invalid argument
```
then I open  the file `Adafruit_Blinka/src/adafruit_blinka/microcontroller/allwinner/h616/pin.py`  ,modified the `PC8 = Pin((1, 72))` to `PC8 = Pin((0, 72))`  . after that, the error disappeared

makermelissa, 2023-08-29:
Would you like to submit a pull request?

sc-bin, 2023-09-05:
Yes,wait a little longer. Our company is preparing to launch a board based on h616. 
the pin.py file of h616 using the general library to control the gpio, But he can't meet our needs 。 We have to created a python library and replace the general library in pin.py 
After our development board is released, if the test has no bugs, we will submit a pull request

closed 2023-09-05 by sc-bin